hrvach/deskhop #99: Middle mouse button sends multiple button presses

techge, 2024-06-08:
I had a strange behavior where pressing the middle mouse button (the one of the wheel) resulting in "going back" in my browser (like using the previous page icon in the browser).

After checking xev with and without deskhop, I realized that deskhop is sending three buttons when using the middle mouse button: 2, 8, 9. When using the very same mouse without deskhop, only button 2 is send (as intended).

Button 8 and 9 is sometimes associated with thumb buttons which are usually used for going back or forward. See this [arch wiki page](https://wiki.archlinux.org/title/Mouse_buttons#Thumb_buttons_-_forward_and_back). This is why my browser behaves the way described above.

So question is... why does deskhop send three buttons instead of just button 2 as it is supposed to (imho)? Can we fix that?

hrvach, 2024-06-09:
Can you put the mouse usb hid descriptor and a few captured reports from both the mouse and the box? Buttons shouldn't be altered in any way, but bugs do happen

techge, 2024-06-10:
I tried to look around a bit in the code to find the cause of this behavior and now I am wondering...

We have a update routine [here](https://github.com/hrvach/deskhop/blob/main/src/mouse.c#L80) where we store the values of a [uint32_t](https://github.com/hrvach/deskhop/blob/main/src/hid_parser.h#L53) in a [int16_t](https://github.com/hrvach/deskhop/blob/main/src/main.h#L261) variable. Honest question as I am not a regular developer: this might be a problem, doesn't it?

techge, 2024-06-11:
Although I am still wondering if the above would be an issue, it might be related to my mouse in the end. Today I wanted to test around again, but could not reproduce the issue. The only difference I could think of is the mouse and keyboard setup I am using right now.

I gonna test the original mouse/keyboard pair end of next week and report my findings.

techge, 2024-06-21:
Okay, the behavior only occurs with a specific mouse I have. Why that is, I can not explain. Anyway, I do not think deskhop need to support any random mouse, there are more interesting topics and said mouse is really a shitty throw-away one, therefore I am closing this issue.

closed 2024-06-21 by techge