hrvach/deskhop #101: Keyboard/Mouse work with 2 active devices, but only one peripheral works with 1 active device

Francommit, 2024-06-09:
Using two devices, the device works perfectly.
As soon as I turn off one of the devices, I've found that I either only get the mouse OR the keyboard functionality.

As soon as I turn on the other device, the device that was previously not working, begins to work.

techge, 2024-06-09:
You may choose a "main" device (either A or B) and put both - mouse and keyboard - on a hub and connect it on the side of this "main" device. Then you can use mouse and keyboard without the other port used.

See also #9

Francommit, 2024-06-09:
Sorry what do you mean by 'connect it on the side' ?

Should I put the USB hub into the keyboard or mouse slot?

Francommit, 2024-06-09:
Just read this - https://github.com/hrvach/deskhop/issues/9#issuecomment-1895703975

Trying...

Francommit, 2024-06-09:
Actually not sure I do understand. What's putting a USB Hub going to do at all?

techge, 2024-06-09:
The thing is that normally the mouse is powered by device B and the keyboard by device A. So when one of the devices is powered off, the responding device is powered down, too. So the deskhop is re-configured so that one pico can power and control both when it was attached with a usb hub. So if you connect the usb hub to device A and connect both - the mouse and the keyboard - to the usb hub, then both are always powered when device A is always connected.

I hope this explanation was of help.

closed 2024-06-12 by Francommit