hrvach/deskhop #105: FPS/Games behaviour

Bunny-Head, 2024-06-17:
Hey!

After receiving it today it's working better than I expected, the "jump" between screens is amazing! Unfortunately, I run out Doom to see how well it worked and I had two issues:

- After trying some other FPS. Games behavious like the sensitivity on the mouse is hyper high.
- After using a minuscule sensitivity, I discover that you can't turn 360º anymore.

BTW, is there a way to customize shortcuts?

Thank you!

hrvach, 2024-06-17:
For now, you can redefine shortcuts in keyboard.c and recompile. Upcoming version (in testing) has a GUI for configuration and the plan is to move the keyboard shortcuts there, so you can redefine it however you want.

I am clueless about games, but I will try making a passthrough / game mode. that prevents weird mouse behaviour with FPS. 

Sorry for the slow progress, work is crazy and personal time is scarce.

Bunny-Head, 2024-06-17:
Thank you so much! I am aware that this projects are made out of passion and it take time to fix thing so please, don't apologize for that - In fact I feel a little useless as I don't know how to do this kind of things and I always want to help the only way I (almost) can: Giving feedback.

Will the GUI be "net" base like other projects that also use Pi like GP2040 arcade PCB project? Just asking out of curiosity.

Again, thank you for your hard work! Have been using it for the first day today and I can tell you already that this is a life changer.

closed 2024-06-18 by Bunny-Head