hrvach/deskhop #107: Ghost wheel scroll

Bunny-Head, 2024-06-18:
Hey! Another little feedback: Noticed that when using Deskhop + USB HUB and scrolling with mouse's wheel, after doing "a scroll" the mouse makes another one after some miliseconds (very noticeable and annoying). Somebody else had this issue? It's a known issue or could be my HUB? It doesn't do that without the Deskhop (checked right now for several minutes).

I can record a video if I didn't explain myself well.

hrvach, 2024-06-18:
Do you perhaps have another mouse at hand to try with? I haven't seen that yet - sounds like delayed packet dequeue, but unsure of the reason behind it.

Bunny-Head, 2024-06-19:
Sorry for the delay on the test, yesterday was office day. Yes, I do have another, more generic way less fancy mouse. Tried it right now and the same issue happens.

Fallingonion, 2024-08-20:
I've just noticed the same thing happening on v0.63, using a Logitech Performance MX mouse: I'm not using the screensaver mode. 
If you're on say a webpage and you scroll with the wheel and then stop, it keeps doing little additional scroll movements. However it stops as soon as you move the mouse.
Hopefully that's helpful :)

jalmeroth, 2024-11-19:
Does this issue persist or can we close this issue?

Bunny-Head, 2024-11-19:
Issue solved.

closed 2024-11-19 by Bunny-Head