hrvach/deskhop #110: WebIf read wrong device values

chris-010, 2024-06-24:
On first attempt when hitting the read button to get device settings, i just get random values.
It's just working on second attempt.

hrvach, 2024-06-24:
All of them or just some?

chris-010, 2024-06-24:
I'm not sure if all, but at least some values as i could notice.

chris-010, 2024-06-26:
First read **attempt**
[image]

Second read **attempt**
[image]

tomjohnstone, 2024-07-04:
I had the same issue when upgrading to the RC. Everytime I would change config settings and reload the config page they would be corrupted. I fixed the issue by wiping my config first.

hrvach, 2024-07-04:
Thanks for noticing this, I have an idea why this might have been happening and reworked that part of the readout code so it should be more stable. 0.62 will be available as soon as I find some free time to update/rebuild/package.

hrvach, 2024-08-03:
This should be resolved, I think. Please let me know if it happens again.

closed 2024-08-03 by hrvach