hrvach/deskhop #111: Keyboard stopped working with 0.60

rantanlan, 2024-06-24:
Hey there, werid issue with 0.60. I updated B as normal, but the config keyboard shortcut didn't work. So I updated the A board too, after that the my keyboard stopped working. Another one is working fine. Any idea? Heres the hiddump of my keyboard (it'S nothing fancy)

`WinHIDdump:
2E8A:107C: Hrvoje Cavrak - DeskHop Switch
PATH:\\?\hid#vid_2e8a&pid_107c&mi_00&col01#7&27318fab&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  01  05  07  19  E0  29  E7  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  08  81  02
  95  38  81  03  05  08  19  01  29  05  95  05  91  02  95  03
  91  03  C1  00
  (52 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_00&col01#8&11ce1c14&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  01  05  07  19  E0  29  E7  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  08  81  02
  19  00  29  97  95  98  81  02  05  08  19  01  29  03  95  03
  91  02  95  05  91  03  C1  00
  (56 bytes)
2E8A:107C: Hrvoje Cavrak - DeskHop Switch
PATH:\\?\hid#vid_2e8a&pid_107c&mi_00&col03#7&27318fab&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  03  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  20  81  03  C1  00
  (28 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_00&col02#8&11ce1c14&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  20  81  03  C1  00
  (28 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_00&col03#8&11ce1c14&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  06  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_01#8&35a559d6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  42  FF  09  01  A1  01  09  01  15  00  25  FF  35  00  45
  00  65  00  55  00  75  08  95  40  81  02  09  01  91  02  C1
  00
  (33 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_02#8&2f4dd45a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  42  FF  09  02  A1  01  09  02  15  00  25  FF  35  00  45
  00  65  00  55  00  75  08  95  40  81  02  C1  00
  (29 bytes)
1B1C:1BAD: Corsair - CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
PATH:\\?\hid#vid_1b1c&pid_1bad&mi_03#8&b769698&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  05  09  01  A1  00  05  09  19  01
  29  20  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  20  81  02  05  01  09  30  15  01  26  FF  7F  45  00  75
  10  95  01  81  06  09  31  81  06  09  38  25  7F  75  08  81
  06  C1  00  C1  00
  (69 bytes)
2E8A:107C: Hrvoje Cavrak - DeskHop Switch
PATH:\\?\hid#vid_2e8a&pid_107c&mi_00&col04#7&27318fab&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}

Unhandled Exception: System.NotSupportedException: Unable to reconstruct the report descriptor.
   at HidSharp.Platform.Windows.WinHidDevice.GetRawReportDescriptor()
   at winhiddump.Program.Main(String[] args)`

hrvach, 2024-06-24:
Does 
[this one](https://github.com/user-attachments/files/15953395/deskhop-boot.zip) fix it? Seems your keyboard is NKRO, it should be correctly parsed:

Keyboard:
   modifier offset: 0, size: 8, report_id: 1
   nkro     offset: 8, size: 152
   
   but something is not as expected. I'll try to check the code once again... apologies.

rantanlan, 2024-06-24:
This fixed it and the keyboard is working again. Thanks! And no need for apologies. I need to thank you for your awesome project!
The config shortcut doesn't seem to work on either keyboards, though.

hrvach, 2024-06-24:
> This fixed it and the keyboard is working again. Thanks! And no need for apologies. I need to thank you for your awesome project! The config shortcut doesn't seem to work on either keyboards, though.

Yeah because I'm a dumbass and put left CTRL + right SHIFT + c + o, because I found a keyboard that can't register both shift pressed at the same time. Updated the binary back so it's consistend and might change in the future but for now it's shifts. Thanks for reporting this, it's super valuable feedback - my own NKRO works, but testing on just 1 device is not really a high-confidence operation. :)

hrvach, 2024-06-24:
@rantanlan if you'd be kind enough to test [deskhop-test2.zip](https://github.com/user-attachments/files/15957042/deskhop-test2.zip) - this might break your keyboard again, but I have a theory what's wrong and I'd like to test it :)

rantanlan, 2024-06-24:
just threw this onto A, keyboard is still working, but the config shortcut seems gone...

hrvach, 2024-06-24:
Ok, that's promising - maybe I actually figured out what was wrong :) the shortcut should probably be changed because of keyboards that can't do dual shift. Looking at you, Lenovo.

closed 2024-06-25 by hrvach