hrvach/deskhop #113: Ubuntu  p22.04 : Build error

OrpheeGT, 2024-06-27:
Hello,

I'm not able to build from source on Ubuntu 22.04.

I first had the following error : 

```
cmake -S . -B build
PICO_SDK_PATH is /home/orphee/Downloads/PCB/deskhop/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
CMake Error at pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
  Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
  "PICO_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake:20 (pico_find_compiler)
  /usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:12 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

Fixed by installing `gcc-arm-none-eabi` package

But now i have the following error : 

```
$ cmake -S . -B build
PICO_SDK_PATH is /home/orphee/Downloads/PCB/deskhop/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/orphee/Downloads/PCB/deskhop/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
TinyUSB available at /home/orphee/Downloads/PCB/deskhop/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/orphee/Downloads/PCB/deskhop/build

$ cmake --build build
Scanning dependencies of target bs2_default
[  1%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  1%] Linking ASM executable bs2_default.elf
[  1%] Built target bs2_default
[  2%] Generating bs2_default.bin
[  2%] Generating bs2_default_padded_checksummed.S
[  2%] Built target bs2_default_padded_checksummed_asm
[... 116 lines cut ...]
[ 32%] Building C object CMakeFiles/board_A.dir/src/usb_descriptors.c.obj
[ 32%] Building C object CMakeFiles/board_A.dir/src/defaults.c.obj
[ 33%] Building C object CMakeFiles/board_A.dir/src/hid_parser.c.obj
[ 33%] Building C object CMakeFiles/board_A.dir/src/utils.c.obj
[ 34%] Building C object CMakeFiles/board_A.dir/src/handlers.c.obj
[ 34%] Building C object CMakeFiles/board_A.dir/src/setup.c.obj
[ 35%] Building C object CMakeFiles/board_A.dir/src/keyboard.c.obj
[ 35%] Building C object CMakeFiles/board_A.dir/src/mouse.c.obj
[ 35%] Building C object CMakeFiles/board_A.dir/src/led.c.obj
[ 36%] Building C object CMakeFiles/board_A.dir/src/uart.c.obj
[ 36%] Building C object CMakeFiles/board_A.dir/src/usb.c.obj
/home/orphee/Downloads/PCB/deskhop/src/usb.c: In function 'tuh_hid_mount_cb':
/home/orphee/Downloads/PCB/deskhop/src/usb.c:135:13: error: a label can only be part of a statement and a declaration is not a statement
  135 |             uint8_t num_parsed
      |             ^~~~~~~
gmake[2]: *** [CMakeFiles/board_A.dir/build.make:216 : CMakeFiles/board_A.dir/src/usb.c.obj] Erreur 1
gmake[1]: *** [CMakeFiles/Makefile2:1588 : CMakeFiles/board_A.dir/all] Erreur 2
gmake: *** [Makefile:91 : all] Erreur 2
```

I updated gcc to version 13 (was 11 at first) but did not fix the issue.

What do I miss ?

Thanks

hrvach, 2024-06-27:
Try git pulling again and rebuilding, moved that variable out of the case block... it's funny, doesn't complain for me.

OrpheeGT, 2024-06-27:
Thanks I will try just now.

another question :
https://github.com/hrvach/deskhop/blob/main/src/user_config.h#L35

How to set combined key ? like SHIFT + CAPS ?

hrvach, 2024-06-27:
Check keyboard.c, there is a hotkey_combo_t struct at the beginning. It's hopefully well documented.

OrpheeGT, 2024-06-27:
Ok,

Replacing https://github.com/hrvach/deskhop/blob/main/src/keyboard.c#L26

by `    {.modifier       = KEYBOARD_MODIFIER_LEFTSHIFT,`

Should do the job if I understood the other combos correctly :)

OrpheeGT, 2024-06-27:
Thanks, it works now ;)

closed 2024-06-27 by OrpheeGT