hrvach/deskhop #114: Problem compiling...

Bunny-Head, 2024-06-28:
After installing CMake & gcc-arm-none-eabi, and after doing cmake -S . -B build on main directory, I get this:

-- Building for: NMake Makefiles
PICO_SDK_PATH is C:/Users/Fali/Downloads/deskhop-main/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
CMake Error at CMakeLists.txt:12 (project):
  Running

   'nmake' '-?'

  failed with:

   no such file or directory

-- Configuring incomplete, errors occurred!

Can somebody help me with the next step please?

chris-010, 2024-07-04:
I did not get it working on windows either, so i moved to linux for compiling.

hrvach, 2024-07-04:
I never tried building it on Windows, so I can't help much there. If anyone has, please share instructions and I'll add them to the wiki. For the time being, I'd advise sticking to Linux.

tomjohnstone, 2024-07-16:
> I never tried building it on Windows, so I can't help much there. If anyone has, please share instructions and I'll add them to the wiki. For the time being, I'd advise sticking to Linux.

Instructions for Windows compilation:

1. Run through setup for WSL (Ubuntu): https://learn.microsoft.com/en-us/windows/wsl/install
2. Open a Linux shell by running `ubuntu` from a Windows command prompt or the run menu
3. Enter the following commands to setup pre-requisites:
```
sudo apt-get update
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib g++
git clone https://github.com/hrvach/deskhop.git
cmake -S . -B build
```
4. Compile code:
```
cmake --build build
```

jalmeroth, 2024-11-19:
Can we close this issue?

Bunny-Head, 2024-11-19:
Solved!

closed 2024-11-19 by Bunny-Head