hrvach/deskhop #118: Logitech MX Master side scroll

cornfeedhobo, 2024-07-02:
I have a logitech plugged in and all the buttons work except the horizontal scroll.

I'll happily tackle this once I can figure out how to build for arm from my local setup.

jalmeroth, 2024-08-17:
Hi @cornfeedhobo,

I started to work around my Logitech related issues in a seperate project over here: https://github.com/jalmeroth/deskhopl.

Would be good to know, if it solves your issue too.

cornfeedhobo, 2024-08-22:
@jalmeroth Interesting. What prevents this from being merged upstream? Why a dedicated fork?

jalmeroth, 2024-08-23:
Hi, see the last posts in this discussion:
https://github.com/hrvach/deskhop/issues/47

FawazAT, 2024-09-30:
Hi, I’m experiencing the same issue with the horizontal scroll. Did you manage to find a solution?

cornfeedhobo, 2024-09-30:
Nope. I'm waiting on @hrvach to have some free time to review @jalmeroth's work.

cornfeedhobo, 2024-11-18:
@Abdulaziz-Alsughaier can you open a PR for this?

jalmeroth, 2024-11-18:
> I haven't properly understood this issue until reading the patch suggested. So, this "side scroll" is actually the horizontal wheel you're missing? Yeah, I can look into this but don't have any means to test so I'd kindly require somebody to try it out afterwards.

In my setup it actually behaves exactly as the wheel field, just that it is the next byte (AC Pan).
The only thing thats different, is that it is using ` HID_USAGE_PAGE_CONSUMER`.

Let me know, if you've something to test

hrvach, 2024-11-18:
> In my setup it actually behaves exactly as the wheel field, just that it is the next byte (AC Pan). The only thing thats different, is that it is using ` HID_USAGE_PAGE_CONSUMER`.
>
> Let me know, if [...]

My only wheel not attached to a car is the vertical wheel on my mouse, so please excuse me asking dumb questions - how does AC Pan differ from horizontal scroll in behavior?

jalmeroth, 2024-11-18:
It's actually the same here: horizontal scrolling is sent via AC Pan field

I can send some reports later if you want

jalmeroth, 2024-11-18:
Here's the descriptor for starters:
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x95, 0x10,        //     Report Count (16)
0x75, 0x01,        //     Report Size (1)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x10,        //     Usage Maximum (0x10)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x16, 0x01, 0x80,  //     Logical Minimum (-32767)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x09, 0x38,        //     Usage (Wheel)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x05, 0x0C,        //     Usage Page (Consumer)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x95, 0x02,        //   Report Count (2)
0x75, 0x10,        //   Report Size (16)
0x15, 0x01,        //   Logical Minimum (1)
0x26, 0xFF, 0x02,  //   Logical Maximum (767)
0x19, 0x01,        //   Usage Minimum (Consumer Control)
0x2A, 0xFF, 0x02,  //   Usage Maximum (0x02FF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x80,        // Usage (Sys Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x04,        //   Report ID (4)
0x95, 0x01,        //   Report Count (1)
0x75, 0x02,        //   Report Size (2)
0x15, 0x01,        //   Logical Minimum (1)
0x25, 0x03,        //   Logical Maximum (3)
0x09, 0x82,        //   Usage (Sys Sleep)
0x09, 0x81,        //   Usage (Sys Power Down)
0x09, 0x83,        //   Usage (Sys Wake Up)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x01,        //   Report Size (1)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x09, 0x9B,        //   Usage (0x9B)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x05,        //   Report Size (5)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 133 bytes
```

A typical report looks like this:
```
020000000000000001
```
Notice the `0x01` which is scrolling horizontally one direction

and here in the other direction `0xff`:
```
0200000000000000ff
```

Here parsed by wireshark:
[image]
[image]

Hope this helps

cornfeedhobo, 2024-11-18:
@hrvach I had some issues compiling the last time I tried, but assuming I can get over that hurdle, yes happy to test!

jalmeroth, 2024-11-18:
Hi @hrvach,

I think I nailed it. :sweat_smile: At least it works on my machine (see [PR#174](https://github.com/hrvach/deskhop/pull/174)).

Let me know what you think,
Jan

cornfeedhobo, 2024-11-19:
Tested and commented on your PR. This works! Thank you so much!

closed 2024-11-19 by hrvach

hrvach, 2024-11-19:
FYI just test-driving it a little before releasing a new version. It's a minor change but being careful is never a bad thing.

PR #174 (2024-11-19): feat: add horizontal scrolling
Fixes #118 (I guess)