hrvach/deskhop #127: Screen lock broken

thawk, 2024-08-02:
After upgrade to v0.61, screen lock function is broken.
I have a Windows and a MacOS computer, when the Windows computer is actived, after `RAlt RShift L` is pressed, Windows computer receives the correct `Win L`, but the MacOS receives the incorrect `Ctrl Alt Q`.
When the MacOS computer is actived, after `RAlt RShift L` is pressed, Windows computer receives the incorrect `Ctrl Alt Q`, the MacOS receives the incorrect `Win L`.
It seems that, #83 and #89 are rollbacked by 1415c1d. Maybe it is the cause?

hrvach, 2024-08-03:
Thanks for catching it, very good diagnostics! Promoting the other branch introduced a few regressions which should be fixed shortly.

hrvach, 2024-08-03:
Should be fixed in v0.62, once again thanks for going the extra mile beyond "it doesn't work"..

closed 2024-08-03 by hrvach