hrvach/deskhop #129: 0.62 & Lock Combination

Bunny-Head, 2024-08-03:
Hi! After updating to 0.62 right now, when I press Control+L, I get the Windows Lock screen and the Computer doesn't get "lock" on it only, I'm still able to jump between the two machines.

hrvach, 2024-08-03:
> Hi! After updating to 0.62 right now, when I press Control+L, I get the Windows Lock screen and the Computer doesn't get "lock" on it only, I'm still able to jump between the two machines.

Ctrl K now, I somehow keep choosing the wrong keyboard shortcuts and various systems have issues with them. :-(

Bunny-Head, 2024-08-04:
Oh, thank you for the clarification! Will try again on 3/4 days as I have to work and Deskhop is now an angular piece of my work setup and can't go back as I was having another couple of weird issues. I said that it was locking the screen (as when you use windows+L on a regular windows computer) which shouldn't be the outcome anyway but it also was opening the Windows Assistant instead radonmly using the same key combination. It also gave me super weird mouse lag and it even stopped working at all (both mouse and keyboard).

It's probably related from a faulty update (which doesn't makes much sense anyway as if it was faulty it shouldn't work at all) but as said, will try in a couple of days and report back.

Million of thanks for your work!

hrvach, 2024-08-04:
That sounds odd, perhaps there is something else happening - I'm interested to learn about any potential issues as stability and usability is the number 1 priority. I've been testing it quite extensively and, at least in my case, it works fairly stable. Let me know if you'll have additional issues. Glad to hear it's useful!

Bunny-Head, 2024-08-18:
I said that I was going to report but funny enough, the next day my laptop explode and haven't been with own PC until today. I started to getting ready when received the notification of the new version and everything is fixed now plus the game mode, unless i find something else, it's working like a charm.

Again, thank you so much for your work!

closed 2024-08-18 by Bunny-Head