hrvach/deskhop #131: Widescreen Support - very sensitive mouse on the X axis

superbreadi, 2024-08-05:
> What happens if I have two different resolutions on my monitors?
> The mouse movement is done in abstract coordinate space and your computer figures out how that corresponds with the physical screen, so it should just work.

Hi, it works, but:

- PC A:  on my DFHD+ monitor (32:10 / 3840 × 1200) I have to set the speed to X = 16 and Y = 51 with acceleration off. Otherwise my Mouse is very fast from left to right, but slow from top to bottom.

- PC B :On my laptop screen (16:10 / 2560 × 1600), I have it set to X = 40 and Y = 45 with acceleration off.

I have a specific use case: when I shut down my second PC, I switch from my laptop screen (16:10) to the larger 32:10 screen. In this setup, my mouse moves extremely fast from left to right but very slowly from top to bottom.

Did I set something up incorrectly?

hrvach, 2024-08-14:
With your settings of different speeds, does it work OK / as expected? 

I have plain 16:9 screens and was never able to clearly see these differences in X vs Y speeds, but I'm not a very demanding mouse user.

Can you please share what OS do you have on PC A and on PC B? Your 51/16 gives 3.19 which is pretty close to 32:10 ratio, but the other one is 1.125 vs 1.6 ... might be due to how OS handles these. Appreciate the data points, if it's OS related we'll fix it by default to 16:9 maybe, with some additional work on the options.

superbreadi, 2024-08-14:
With the speed settings it works as expected.
On PC A i have an linux os and on PC B windows. But on the widescreen there is no difference between windows or linux regards to the speed settings. 51/16 is the perfect setting.
On PC B (windows laptop) i fine tuned my settings to x=33 y=52 wich is close to the 1.6 factor...

In windows powershell the absolute mouse position you get with:
```
x*(65535/System.Windows.Forms.Screen.PrimaryScreen.Bounds.Width);
y*(65535/System.Windows.Forms.Screen.PrimaryScreen.Bounds.Height);
```
so in my case:
PC A:
65535/3840 = 17.06
65535/1200 = 54.6125
PC B:
65535/2560 = 25.599609375
65535/1600 = 40.959375
But on PC B i have 125% windows screen scaling:
25.599 * 1,25 = 31,9995 ≈ 32
41 * 1,25 =51,1992 ≈ 51
which resemble my fine tuned settings reallly close 👍

hrvach, 2024-08-18:
Thank you for the clarification, the latest update has changed defaults to accommodate the most common display geometry (16:9), the rest should be configurable through the UI to best fit user's preferences.

closed 2024-08-18 by hrvach