hrvach/deskhop #132: 0.62 Caps Lock not switching board

troed, 2024-08-09:
I've previously been using <0.6 for a long time without issues, and today I upgraded to 0.62. Everything's working fine, both boards blink, I can use the (very nice!) new web configuration and I can also switch between computers by moving the mouse.

But nothing happens when I click Caps Lock. Or, well, IT WORKS AS CAPS LOCK which is just horrible :D

Is this no longer the default, and if so, which key combination is?

Fallingonion, 2024-08-09:
I believe newer versions have it effectively disabled by setting it to an "impossible" value, function key 24.
You can set it back to caps lock in config.h as before :)

hrvach, 2024-08-14:
Yes, some folks complained they wanted their caps lock back ... so, what would be a good solution here in your opinion?

Caps lock works as normal, ctrl + caps switches? 
Or caps switches and ctrl + caps lock for actual caps lock?

What's the most practical option here?

Fallingonion, 2024-08-14:
For me ctrl+caps to switch would be preferable as I rarely use the keyboard to switch but definitely use caps lock...

troed, 2024-08-14:
The current documentation doesn't make it very clear that it's not in the firmware by default :) That's what caused me to post the issue. I'm ok with either having to manually add it, it being configurable or some weird key combination that otherwise isn't used. Like ctrl+caps.

hrvach, 2024-08-18:
Like the saying goes, "the people are always right"! :-) I've updated documentation, and switched to using ctrl + caps as suggested in v0.63. Hope this makes it more convenient for everyone, feel free to reopen the issue if not satisfactory. Thank you for pointing it out!

closed 2024-08-18 by hrvach