hrvach/deskhop #133: Mouse Cherry MW 8C Advanced doesn't work (older version worked fine)

chri2, 2024-08-10:
The battery went flat in my **Cherry MW 8 Advanced** mouse and I replaced it with - as I thought - the same model, but now with usb-c connector to charge: **Cherry MW 8C Advanced**. The new mouse does not work.

Connected to the deskhop the newer model just makes the pointer jump around, transmits keycodes when moved and does send multiple button events when buttons are pressed.

## old/works: Cherry MW 8 Advanced
lsusb: Bus 001 Device 023: ID 046a:0114 CHERRY CHERRY Wireless Device
P/N: JW-8000
M/N: JF-80

```
$ sudo usbhid-dump -a 1:23
001:023:001:DESCRIPTOR         1723289925.052961
 05 01 09 02 A1 01 85 03 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 75 01 95 05 81 02 75 03 95 01
 81 01 05 01 09 30 09 31 16 01 F8 26 FF 07 75 0C
 95 02 81 06 09 38 15 81 25 7F 75 08 95 01 81 06
 C0 05 0C 15 81 25 7F 75 08 95 01 0A 38 02 81 06
 C0 05 0C 09 01 A1 01 85 01 19 00 2A 80 03 15 00
 26 80 03 75 10 95 01 81 00 C0 05 01 09 80 A1 01
 85 02 19 81 29 83 25 01 75 01 95 03 81 02 95 05
 81 01 C0 06 BC FF 09 88 A1 01 85 04 19 00 2A FF
 00 15 00 26 FF 00 75 08 95 01 81 00 C0 06 F2 F1
 0A F2 F1 A1 01 85 05 09 00 95 06 75 08 15 00 26
 FF 00 91 02 09 00 95 05 75 08 15 00 26 FF 00 81
 02 C0

001:023:000:DESCRIPTOR         1723289925.055604
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 75 08 95 01 81 01 75 01 95 03
 05 08 19 01 29 03 91 02 75 01 95 05 91 01 75 08
 95 06 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```

## new/doesn't work: Cherry MW 8C Advanced
lsusb: Bus 001 Device 019: ID 046a:c117 CHERRY CHERRY Wireless Device
P/N: JW-810000
M/N: JF-81

```
$ sudo usbhid-dump -a 1:24
001:024:002:DESCRIPTOR         1723289986.892455
 05 0C 09 01 A1 01 85 01 05 0C 09 01 A1 01 75 10
 95 01 15 01 26 9C 02 19 01 2A 9C 02 81 00 C0 05
 01 09 80 A1 01 09 82 09 82 09 83 15 00 25 01 75
 01 95 03 81 02 75 01 95 05 81 01 C0 06 BC FF 09
 88 A1 01 75 08 95 02 15 00 26 FF 00 19 01 2A FF
 00 81 00 C0 C0

001:024:001:DESCRIPTOR         1723289986.894809
 05 01 09 02 A1 01 85 01 09 01 A1 00 95 05 75 01
 05 09 19 01 29 05 15 00 25 01 81 02 95 01 75 03
 81 01 75 08 95 01 05 01 09 38 15 81 25 7F 81 06
 05 0C 0A 38 02 95 01 81 06 C0 85 02 09 01 A1 00
 75 0C 95 02 05 01 09 30 09 31 16 01 F8 26 FF 07
 81 06 C0 C0 06 A0 FF 09 01 A1 01 85 09 09 01 15
 00 26 FF 00 75 08 95 07 91 82 09 02 75 08 95 07
 81 82 C0

001:024:000:DESCRIPTOR         1723289986.896453
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```

chri2, 2024-08-10:
I wanted to repair the older model anyway and to be able to work I moved the battery from the new version into the old one until the replacement battery arrives.

But still I'd like to use the usb-c version (newer model) in my second workplace.

hrvach, 2024-08-10:
I think I know what's going on here, X/Y movements have a different report ID and are part of a different collection. I've never seen this before, will check to see if it could be addressed in a future release...

chri2, 2024-08-10:
Thanks a lot! Would be really great if it'd work since I like that mouse and got used to it (and just bought two new ones believing they would be the same).

hrvach, 2024-08-14:
I have a general idea what needs to be done, just need to find a few free hours to do it (which is impossible lately)... appreciate your patience.

jalmeroth, 2024-11-19:
Does this issue persist or can we close this issue?

chri2, 2024-11-23:
Didn't check lately and am afk. Will test next week and report.

chri2, 2024-11-28:
Issue persists in 0.68 . Mouse movement still results partially in keyboard input. Deskhop is not usable with this mouse.

chri2, 2025-01-23:
Just installed 0.70.0 and the issue still persists.

Is there anything I could do to help solve this - e.g. send a mouse for testing 😉?

Otherwise I'd have to look for different mice to make it compatible to deskhop again...

chri2, 2025-03-04:
> different mice to make it compatible to deskhop

Replaced my two mice with the older working micro-USB-version.

> e.g. send a mouse for testing

A not-yet-working usb-c mouse sits on my desks and I could send it to you, @hrvach, for debugging to keep it if you like it.

chri2, 2025-03-18:
Fixed in 0.71 👍🥳❤ .

hrvach, 2025-03-18:
> Fixed in 0.71 👍🥳❤ .

Some more bugfixes coming soon, but the past few months have been insane work-wise. Having high hopes spring will be better and I can finally push some new features out! :)

chri2, 2025-03-18:
like https://github.com/hrvach/deskhop/issues/217 🙃 ?