hrvach/deskhop #134: Linux + VMware mouse problems

chri2, 2024-08-12:
On my notebook I'm using Debian Bookworm with an additional external display connected. When I'm running VMware remote console 12.0.5 build-22744838 for a Windows VM I'm having problems using the mouse. The same mouse connected directly to the notebook works just fine.

Problems observed:
* No mouse pointer shown inside the console when focus taken by the console
* When clicking to put input focus inside the remote console the guests mouse pointer freezes in position even though it seems I can still move an invisible mouse pointer and click on buttons if I'm able to guess the right mouse position
* Starting to type inside an guest freezes the mouse pointer. Releasing the input focus by `ctrl`-`alt` and activating the input to the guest by clicking makes the guests mouse pointer work again until I activate an input field and type some stuff.

hrvach, 2024-08-14:
I haven't tried vmware before, but I reckon you should be able to work around this with "direct mode" that should soon be out in the next release... tl;dr - a key combo that locks the mouse to the current screen and turns it into a plain, relative mouse. Would this be a reasonable solution?

chri2, 2024-08-15:
That would be a great solution. Sounds simple and effective (not increasing code complexity a lot I hope). I'll try when the feature is available.

hrvach, 2024-08-18:
@chri2 implemented in v0.63, please check and let me know if this works with your VM. Cheers!

jalmeroth, 2024-11-19:
Does this issue persist or can we close this issue?

hrvach, 2024-11-19:
Gaming mode has been integrated in the recent releases, I think that should make virtualization at least "usable". Closing.

closed 2024-11-19 by hrvach