hrvach/deskhop #136: Configuration interface not working

github-k8n, 2024-08-13:
Updated both RPis using the "hold button on pi" method and tried with both self-compiled uf2 as well as 0.62 prebuilt image.
Things seem to work, it still switches back and forth between the monitors, but it does not react at all to the configuration "Left Shift + Right Shift + C + O" , no reboot, no USB mass storage device, no blinking...

hrvach, 2024-08-17:
Maybe the keyboard used is unable to send both shifts at the same time? If you have the self-compiled one, perhaps try changing the shortcut (in keyboard.c) to e.g. left control + right shift + c + o and rebuilding? Let me know if it still doesn't work.

SeanDennis101, 2024-08-20:
I Probably need to compile instead of using the prebuilt image because I am having the same issue as the other commenter.  Will try a different keyboard first to make sure.  The post comment from last week from github-k8n describes it exactly basically same exact issue.

github-k8n, 2024-08-22:
Worked after I changed it to right Ctrl + right Shift + C + O

FYI the keyboards involved are
- IBM Model M (connected using a PS/2 to USB adapter "ID 04d9:1400 Holtek Semiconductor, Inc. PS/2 keyboard + mouse controller"
- Microsoft Wired Keyboard 600 (just noticed the third keyboard i tried is also the same just different color... :D )

Neither of those worked. Not sure if detecting/sending both at the same time is something that only recent keyboards can do but maybe another combination might be better for the default setting...

jalmeroth, 2024-11-19:
Does this issue persist or can we close this issue?

github-k8n, 2024-11-19:
Sorry, yes the new default key combo works for me... thanks, can be closed.

jalmeroth, 2024-11-19:
Cool, thanks for confirming! You can close it yourself or @hrvach has to do it 😉

closed 2024-11-19 by github-k8n