hrvach/deskhop #137: Caps lock / num lock LEDs only work on output A

Fallingonion, 2024-08-20:
Hi all... a small cosmetic issue on v0.63, possibly on earlier versions also?
On my keyboard (a Logitech G815) the caps lock and num lock LEDs only work on output A, but not on output B. 
Is this as expected, or am I doing something stupid? I can't remember if it worked on both on previous versions...

Thanks :)

hrvach, 2024-08-23:
Good catch, I think I know what might be going on. Will try to fix in the next release.

hrvach, 2024-08-24:
Please check the pre-release (v0.64), better now?

Fallingonion, 2024-08-24:
Caps lock works correctly now, thanks!  I can't test num lock since the second machine is a laptop with no numpad, so doesn't have a numlock key, but I assume it works :) 
Thanks!

hrvach, 2024-08-24:
Really appreciate you catching this one, not trivial to notice and I would have never fixed it otherwise. Thanks for helping the project with bug reports!

closed 2024-08-24 by hrvach

Fallingonion, 2024-08-24:
Thank *you* for fixing things so quickly!