hrvach/deskhop #139: [pcb] 💡expose uart1 tx pin on swd port

jalmeroth, 2024-08-21:
Hey there,

my current development setup looks something like this:
[image]

Since I have the picoprobes attached, I would like to have the stdio output on uart1 instead of using the deskhop CDC device.
Therefore I would like to have an uart1 tx pin (e.g. GPIO 4) next to the SWD connectors, so I can get rid of the floating, grey cable in the picture.

Would that be possible?

Thank you very much!

hrvach, 2024-08-24:
By all means, I need to look into pcb 1.2 soon anyway, have some outstanding bugs. Also, pico2 is out and that needs to be tested 😃 cool picoprobes, btw have you ever experienced picoprobe port just going bad after a few weeks?

jalmeroth, 2024-08-24:
> By all means, I need to look into pcb 1.2 soon anyway, have some outstanding bugs. Also, pico2 is out and that needs to be tested 😃

Awesome, I was just about to order some new pico2`s!

> btw have you ever experienced picoprobe port just going bad after a few weeks?

No not really, but I must confess that I basically use them as tty and occasionally to update the firmware via openocd. Can you narrow what "going bad" means?

hrvach, 2024-08-24:
> Awesome, I was just about to order some new pico2`s!
>
>
> No not really, but I must confess that I basically use them as tty and occasionally to update the firmware via openocd. Can you narrow what [...]

Yeah, I've had openocd firmware upgrade stop working for some reason on 2-3 boards, but when I recompiled to use other GPIOs, it started working again for a while, only to repeat. Perhaps some TVS diodes might be a good thing to install there :)