hrvach/deskhop #141: Keyboard not switching desktops

Omgmydogisdead, 2024-08-22:
My deskhop is working for the mouse but the keyboard is not switching desktops when the mouse moves to another desktop. This time, it's definitely not a hardware problem as I have re-checked it all. When the keyboard is plugged into board A(the default), the keyboard will only work on desktop A and never desktop B but when the keyboard is plugged into Board B, the keyboard will only work on desktop B and never desktop A. The mouse works normally in both cases.

Omgmydogisdead, 2024-08-22:
yes, I have tried control +  CAPSLOCK. it stopped typing on Desktop A but still didn't start typing on Desktop B

hrvach, 2024-08-22:
This is super weird and I must admit never seeing anything like this. Could you please try with another keyboard first, to rule out the keyboard being a factor here? Please describe your setup if possible. If keyboard works on either computer, that means it's being understood and parsed correctly, and if the mouse switches outputs, it means serial connection should be working. 

[image]