hrvach/deskhop #142: TMK keyboard not relaying input to the connected computers

dsemi, 2024-08-23:
I have a Leopold FC660C running [TMK](https://github.com/tmk/tmk_keyboard) that doesn't appear to be working with the deskhop.  
When plugging in I see the LED flash twice, but no keyboard output ever appears to reach the host computer. Hotkeys for opening up configuration, etc. don't appear to work either.

I can verify that the deskhop itself is working; a different keyboard I tried is working as expected. I also tried compiling the firmware with ENFORCE_PORTS, that didn't make a difference.

I'm assuming the issue is related to the keyboard firmware, would it be possible to add support? Please let me know if there's information I can dump from the board, usb events, etc. that would be helpful in diagnosing the issue.
Somewhat related, is there a more detailed guide on debugging? I'm not familiar with how one would attach a debugger, see log messages, etc. for the pico firmware. If that's easy enough to do then I could be more active in diagnosing why the keyboard isn't functional.

hrvach, 2024-08-23:
Sure, can you please dump the descriptor using usbhid-dump -e descriptor  on Linux? I'm interested in extending support for various keyboards and learning what case I'm not handling properly is essential to that effort. Thanks!

dsemi, 2024-08-23:
Sure, here you go:

```
$ sudo usbhid-dump -e descriptor
001:012:004:DESCRIPTOR         1724451626.060660
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 05 08 19 01 29 05 95 05 75 01
 91 02 95 01 75 03 91 01 05 07 19 00 29 F7 15 00
 25 01 95 F8 75 01 81 02 C0

001:012:003:DESCRIPTOR         1724451626.068659
 06 31 FF 09 74 A1 01 09 75 15 00 26 FF 00 95 20
 75 08 81 02 09 76 15 00 26 FF 00 95 20 75 08 91
 02 C0

001:012:002:DESCRIPTOR         1724451626.078663
 05 01 09 80 A1 01 85 02 16 01 00 26 37 00 1A 81
 00 2A B7 00 75 10 95 01 81 00 C0 05 0C 09 01 A1
 01 85 03 16 01 00 26 9C 02 1A 01 00 2A 9C 02 75
 10 95 01 81 00 C0

001:012:001:DESCRIPTOR         1724451626.091655
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01
 05 01 09 30 09 31 15 81 25 7F 95 02 75 08 81 06
 09 38 15 81 25 7F 95 01 75 08 81 06 05 0C 0A 38
 02 15 81 25 7F 95 01 75 08 81 06 C0 C0

001:012:000:DESCRIPTOR         1724451626.102658
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 05 08 19 01
 29 05 95 05 75 01 91 02 95 01 75 03 91 01 05 07
 19 00 29 FF 15 00 26 FF 00 95 06 75 08 81 00 C0

```

hrvach, 2024-08-23:
I think I know what it is, there was another report with the same root cause - NKRO with no report ID. That is an unsupported case I need to fix, somehow strangely assumed NKRO would always come with report ID. 

The fix shouldn't be super complicated, maybe removing:

```
     if (report_id > 0
        && report_id == iface->keyboard.nkro.report_id
```
and leaving just if (iface->keyboard.is_nkro) in hid_report.c might work here.

dsemi, 2024-08-23:
Unfortunately that doesn't seem to have helped.

Should there be that many descriptors in the output? I unplugged everything except the keyboard before running `usbhid-dump` and there were still 5 descriptors.
Could it be something like what you mentioned [here](https://github.com/hrvach/deskhop/issues/71#issuecomment-2016583247)?

hrvach, 2024-08-24:
You could try checking how to disable nkro on tmk but I would prefer to actually fix it, give me a bit to look into the code and try a few things first

hrvach, 2024-08-24:
Any chance you might be able to try [this one](https://github.com/user-attachments/files/16737175/deskhop.zip) for me? Don't have a similar keyboard so it's hard for me to recreate, so I'm just guessing what's happening but curious if this would fix the issue you're experiencing. Thanks!

dsemi, 2024-08-24:
That works! I had a couple times where I started typing and after a few keys it seemed to get stuck again, but unplugging and re-plugging the keyboard seems to have at least fixed it for now; I can't reproduce anymore.

Thanks! What was the change that did it?

hrvach, 2024-08-24:
Hopefully no keyboard replugging should be needed, please test a bit more and let me know. The case with no report ID and ITF protocol NONE was not covered for some reason, I'll be releasing a fix soon...

dsemi, 2024-08-24:
Yeah, something is still a bit off. It was working well for a bit connected only to one computer, but once I connected the other and added the mouse, I could consistently get to a state where the keyboard input stops working.  

I noticed that when I stopped moving the mouse, the deskhop would regularly trigger the LEDs back and forth between A and B, like it wasn't sure which computer to output to? It didn't seem to affect the behavior of the mouse, but it lags keyboard input to B during that alternation. And eventually it would hit a state where the keyboard stops working altogether and I have to reset. Haven't found a consistent repro for that fully non-working state, but I'll play around with it some more.

A couple other things I noticed:
- The shortcut for opening the web configuration is not working (left shift + right shift + c + o). Not sure if that's an issue with the firmware or if the keyboard just doesn't like sending both shifts. I'd try remapping it, but I'm not sure what changes you made, so I can't try recompiling with a different shortcut, and I can't get the web config to open to remap it
- Other keyboard shortcuts appear to be working. I tried right ctrl + k and the mouse locks to the current screen as expected. Is it intentional for the keyboard shortcuts to pass through keys to the computer? When I press ctrl + k it locks to output, but still sends a 'k' that I can see if I have a text field focused when I do it. Not sure if this is specific to this keyboard or just a general bug

hrvach, 2024-08-24:
With latest rev 0.63 it was changed to left ctrl + right shift + c + o because a few people complained their keyboards can't send both shifts - that was a bad choice to begin with.

Passthrough for shortcuts can be explicitly specified, for ctrl + K it should be disabled, I'll check. If your LEDs regularly go back and forth, device is probably rebooting. Maybe excessive current, or hitting some weird bug. Does it happen with a different keyboard as well?

dsemi, 2024-08-24:
> With latest rev 0.63 it was changed to left ctrl + right shift + c + o because a few people complained their keyboards can't send both shifts - that was a bad choice to begin with.

Ah, I missed that. Ok can confirm that is working fine then.

>  If your LEDs regularly go back and forth, device is probably rebooting. Maybe excessive current, or hitting some weird bug. Does it happen with a different keyboard as well?

Yes it does happen with a different keyboard. I don't think it's excessive current, it actually only happens when the mouse is plugged in. Even if no keyboard is connected and just the mouse it's doing that. This is also present in 0.63, so maybe a separate issue with my mouse.

hrvach, 2024-08-24:
Changing shortcuts is super bad, I apologise for the confusion, all my keyboards can send both shifts so only later I realised some can't. Pre-release 0.64 is published with a few fixes, thank you for reporting this kbd issue, I somehow missed covering this case.

dsemi, 2024-08-24:
Thanks for the quick fixes! I'll mark this as fixed and file a separate issue for the mouse as that seems to be unrelated to the keyboard.

closed 2024-08-24 by dsemi