hrvach/deskhop #143: Deskhop flickers output LEDs (I think it's resetting?) without constant mouse movement

dsemi, 2024-08-24:
The device appears to be working correctly otherwise, mouse and keyboard input are being sent to outputs A and B where appropriate. But if I stop moving the mouse, after a few seconds the device alternates lighting up the A and B output LEDs, then pauses for a couple seconds, then does it again, and continues over and over.  If I start moving the mouse again, the LED acts as normal and lights up for the correct output.

I think the deskhop is crashlooping? At the very least this interrupts keyboard input while LEDs are flip-flopping.

This issue is mouse-specific, I can only reproduce this with a Logitech G Pro X Superlight 2. I also tested with a MX Master 3 and that doesn't reproduce the bug.

Descriptor dump for the the Pro X mouse:
```
$ sudo usbhid-dump -e descriptor
001:008:002:DESCRIPTOR         1724528938.883045
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0

001:008:001:DESCRIPTOR         1724528938.888040
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 95 05 05 08 19 01 29 05
 91 02 95 01 75 03 91 03 95 70 75 01 05 07 19 04
 29 73 81 02 95 05 19 87 29 8B 81 02 95 03 19 90
 29 92 81 02 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 06 81 03 C0

001:008:000:DESCRIPTOR         1724528938.893041
 05 01 09 02 A1 01 09 01 A1 00 95 10 75 01 15 00
 25 01 05 09 19 01 29 10 81 02 95 02 75 10 16 01
 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01 75 08
 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38 02 81
 06 C0 06 00 FF 09 F1 75 08 95 05 15 00 26 FF 00
 81 00 C0
```

Tested with 0.64 prerelease

hrvach, 2024-08-24:
it would be great if you could usbhid-dump -e stream of both moving the mouse a little and leaving it stationary, not promising I'll find the problem but I'll try...

dsemi, 2024-08-24:
Here you go: [stream-dump.txt](https://github.com/user-attachments/files/16737916/stream-dump.txt)

Most of the output is from moving the mouse around. The last 4 messages I saw ticking once every few seconds, and that appeared to be when the device would flicker.

dsemi, 2024-09-02:
Have you had a chance to take a look at the stream log? I would guess it's some unrecognized heartbeat message that's causing a crash, but I haven't dug around the code to figure out where that could be occurring.  
Something like the suggestion in https://github.com/hrvach/deskhop/issues/145 would likely make this easier to diagnose.

hrvach, 2024-09-03:
Try doing this - increase watchdog timer to 5 seconds, rebuild and see if the blinking interval increases to every 5 seconds. Nothing seems very unusual here, maybe the tinyusb host loop gets stuck for some reason with this device. How does this connect to the computer normally, something like this?

[image]

dsemi, 2024-09-03:
It's not a universal receiver, but a similar dongle specifically for the mouse: 
[image]

I tried updating `WATCHDOG_TIMEOUT` in src/include/main.h to 5000, but didn't notice a difference. The device flickers the LEDs back and forth between outputs about every 2s.

dsemi, 2024-09-04:
Tried with a usb hub, but I'm just not getting any input to the computer then, keyboard or mouse.

hrvach, 2024-09-13:
I tried using exactly the same descriptor and did not see reboots locally. No idea what's causing it, but I'll try making a few changes and will kindly ask you to test and report if that changes anything (i.e. shooting in the dark)

dsemi, 2024-09-13:
Thanks for looking, will be ready to test.

Random idea, maybe it's not actually resetting, just re-triggering the function that runs when it detects a mouse getting plugged in? Perhaps from a low power mode in the mouse where the deskhop thinks the mouse is disconnecting and reconnecting? That would also explain why it doesn't happen while I'm moving the mouse around, and only after I let it sit for a few seconds.