hrvach/deskhop #148: Mouse movement duplicated on second monitor

github-k8n, 2024-09-05:
With my current settings (see below) it seems that sometimes my mouse movement on the right monitor (Linux, Output A) is duplicated on the left one (Windows, 2 monitors, one builtin and one connected, the connected monitor is exactly the same as the one from the Linux system.
Only mouse movement is duplicated not any clicks or keyboard entries. I can fix it by moving the mouse all way to the left (over the connected monitor of the Windows system and on to the builtin monitor) and then back again to the Linux system then everything is ok again for a while.
I have a suspicion that it could be related to either the Cursor park position or the Screensaver mode.

[image]

hrvach, 2024-09-07:
Thank you for a good problem description! If you reconfigure the cursor park position to something else, does that seem to change the behaviour?

github-k8n, 2024-09-07:
Cursor park position does not affect it (set to Top), it still mirrors the movement. However when I set the Screensaver mode from Jitter to Pong, it does no longer cause the same "mirrored mouse" behavior (obviously the mouse will now move in the "pong" zigzag movement on the left (B, Windows) screen which is expected.

=====

I also noticed another behavior, maybe I should open another case for that (let me know and I'll copy/paste it) but wanted to mention it here as well.
(Again, the physical screen layout from left to right:   (B: Builtin Windows, B: External Windows, A: External Linux)

When moving the mouse cursor to the left until it reaches the Builtin Windows screen, it seems the acceleration is suddenly no longer working. Cursor becomes slow (non-accelerated) even when back on the External Windows screen. It only goes back to normal once it reaches the A: External Linux screen. As long as I keep the mouse on the External screens (B: External Windows, A: External Linux) the mouse acceleration will work fine. (maybe some issue with the dual screen stuff on Windows)?

github-k8n, 2024-09-07:
Oh and just because I didn't mention it earlier: it seems to start (the mirrored mouse movement) reproducibly after the screensaver timeout (when using "Jitter").

hrvach, 2024-09-10:
I am struggling to reproduce this, but I'm convinced you get sort of a "split brain" scenario. Next release should have a fix for that, but I'm still unsure of the scenario how it might happen.

github-k8n, 2024-11-19:
Issue still persists with latest release. While luckily only the movement is duplicated, not the clicks, it is highly distracting to have that movement on the other screen, possibly causing confusion as to which screen is actually being controlled.
Again, this only happens with "jitter" screensaver mode, "pong" is not affected.

I'd be happy to test any test firmware you can provide me with, i also have the debug pins populated.
If you tell me what exactly is being outputted (protocol etc) maybe I can get some output there.
I have an FTDI232 usb to TTL adapter, an rs485 to ttl adapter and a bunch of ESPs, or let me know what exactly I would need... (not really sure how to connect to the GND/DATA/CLK connector)

github-k8n, 2024-11-20:
I think there might be some issue with the code?  (the || seems to be part of something else that was cut off... )
Did you maybe mean the following?

```
    case JITTER:

            if (report.x + jitter < MIN_SCREEN_COORD) {
                report.x ++;
            } else if ( report.x + jitter > MAX_SCREEN_COORD) {
                report.x --;
            }

            report.x += jitter;
            jitter = -jitter;
            break;
```

github-k8n, 2024-11-20:
Is jitter always  +1 or -1 ?
Otherwise maybe you can just return report.x instead of first reducing it and adding jitter agin.
```
report.x = (report.x + jitter < MIN_SCREEN_COORD) ? report.x : report.x + jitter;
report.x = (report.x + jitter > MAX_SCREEN_COORD) ? report.x : report.x + jitter;
jitter = -jitter;
break;
```

github-k8n, 2024-11-20:
And the issue still happens ... :(

hrvach, 2024-11-27:
Can you remove all code in the jitter section, just leave the break. Still happens? I'm struggling to recreate. Both boxes think they are masters, but not sure how that would happen using a screensaver.

eltariel, 2024-11-29:
I'm also seeing this behaviour today. Not sure if I've seen it before.

I'm running the same physical layout but both hosts are on MacOS. I'm not sure what the screen saver mode is on this particular device but will look into it some more when I'm done at $work for the day.

hrvach, 2024-12-03:
Any new findings @eltariel ? I have never noticed this yet, but would really like to squash that bug if possible because it seems quite annoying for the user.

eltariel, 2024-12-04:
Haven't had a chance to dig any deeper, but some additional details:

- turning jitter off made the issue stop
- doesn't happen in pong screensaver mode but that's too distracting for me to use
- mouse movements when B is active never get duplicated on A
- unplugging A when B is duplicating breaks it even more: it still follows mouse movements but it won't switch screen and the buttons do nothing. I only tried this once 🤣

eltariel, 2024-12-04:
Think I've figured it out...

When a mouse report is received, we update `state->position_x/y` from the report values. This goes into global state *on the receiving device* - in most cases that's B - before it gets output either directly to a report or via the UART. [link](https://github.com/hrvach/deskhop/blob/cbb98e4a88e8123004405bc3f0fc6b5be7f2cc3b/src/mouse.c#L79)

The jitter screensaver works by alternately adding or subtracting 1 from `state->position_x/y` - *which is set on every incoming mouse report!*

I was going to suggest we might want to just swap the jitter implementation for a relative report instead of the absolute one, but now I'm pretty sure it'll fix it. I'll take a stab at it tonight if I find a moment.

hrvach, 2024-12-04:
@github-k8n I finally managed to recreate it, it's clear what's happening now. And yes - I can see how super annoying it is. :)  

@eltariel tough one, but you figured it out! Even while being an inactive output, pointer_x keeps getting updated locally if the output is idle and screensaver active resulting in a sort of bias to base position the jitter moves around, and the bias value is derived form current mouse position. Fun :)

Relative indeed fixes it but now I've spotted another issue - mouse seems to have a non-zero net movement (to the left in my case), slowly progressing towards the other side. It also wiggles a lot, let me pull in your pr and provide some additional tweaks. Once again, thank you!

PR #181 (2024-12-04): Use relative mouse reports for jitter screensaver mode
Fixes #148.
closed 2024-12-04 by hrvach