hrvach/deskhop #150: Media key regression from 0.63 to 0.64

alankeny, 2024-09-06:
I need to report a regression between release 0.63 and 0.64.  On July 22nd I reported in the "Tester Wanted" discussion that everything was was working with my setup except the media keys on the keyboard including mute, volume up, and volume down.  Shortly after that you made a change, and all my media keys started working with the deskhop-test2.zip version posted in that thread.  They continued to work properly through the 0.63 release.

With the 0.64 release my media keys are back to the previous behavior.  They result in mouse movement and sometimes right-clicks instead of sending mute, volume up, and volume down.  Please let me know if you need me to collect any information from the 0.64 release.  For now I've gone back to 0.63.

Thanks so much for continuing to develop this project!

hrvach, 2024-09-07:
That's odd because no part of the code that touches those was changed. Can you perhaps do usb hid descriptor dump and paste here, I'll try to check what the code might be getting confused with...

alankeny, 2024-09-07:
I'm not sure which one you need, so I'm sending three descriptor dumps:
 * Unifying Receiver directly connected
 * DeskHop switch running 0.63
 * DeskHop switch running 0.64

Let me know if you need anything else.

-----

Directly connected:

    Bus 005 Device 012: ID 046d:c52b Logitech, Inc. Unifying Receiver

    005:012:002:DESCRIPTOR         1725715281.603731
     06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
     FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
     A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
     00 09 02 91 00 C0 06 00 FF 09 04 A1 01 85 20 95
     0E 75 08 15 00 26 FF 00 09 41 81 00 09 41 91 00
     85 21 95 1F 09 42 81 00 09 42 91 00 C0
   
    005:012:001:DESCRIPTOR         1725715281.608095
     05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
     15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 0C
     16 01 F8 26 FF 07 05 01 09 30 09 31 81 06 95 01
     75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
     02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
     10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
     09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
     09 81 09 83 81 00 75 06 81 03 C0 06 BC FF 09 88
     A1 01 85 08 95 01 75 08 15 01 26 FF 00 19 01 29
     FF 81 00 C0
   
    005:012:000:DESCRIPTOR         1725715281.611219
     05 01 09 06 A1 01 95 08 75 01 15 00 25 01 05 07
     19 E0 29 E7 81 02 81 03 95 05 05 08 19 01 29 05
     91 02 95 01 75 03 91 01 95 06 75 08 15 00 26 FF
     00 05 07 19 00 2A FF 00 81 00 C0

-----

DeskHop switch running 0.63

    Bus 005 Device 005: ID 2e8a:107c Hrvoje Cavrak DeskHop Switch

    005:005:001:DESCRIPTOR         1725715573.623076
     05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
     29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
     81 01 05 01 09 30 09 31 16 00 80 26 FF 7F 75 10
     95 02 81 06 09 38 15 81 25 7F 95 01 75 08 81 06
     95 01 75 08 81 01 C0 C0

    005:005:000:DESCRIPTOR         1725715573.626035
     05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
     25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
     19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
     05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
     81 00 C0 05 01 09 02 A1 01 85 02 09 01 A1 00 05
     09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
     01 75 03 81 01 05 01 09 30 09 31 15 00 26 FF 7F
     75 10 95 02 81 02 09 38 15 81 25 7F 95 01 75 08
     81 06 95 01 75 08 81 01 C0 C0 05 0C 09 01 A1 01
     85 03 15 00 26 FF 0F 19 00 2A FF 0F 75 10 95 02
     81 00 C0 05 01 09 80 A1 01 85 04 15 01 25 03 95
     01 75 02 09 81 09 82 09 83 81 00 95 01 75 06 81
     01 C0

------

DeskHop switch running 0.64

    Bus 005 Device 014: ID 2e8a:107c Hrvoje Cavrak DeskHop Switch
    
    005:014:001:DESCRIPTOR         1725715735.517874
     05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
     29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
     81 01 05 01 09 30 09 31 16 00 80 26 FF 7F 75 10
     95 02 81 06 09 38 15 81 25 7F 95 01 75 08 81 06
     95 01 75 08 81 01 C0 C0
    
    005:014:000:DESCRIPTOR         1725715735.520854
     05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
     25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
     19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
     05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
     81 00 C0 05 01 09 02 A1 01 85 02 09 01 A1 00 05
     09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
     01 75 03 81 01 05 01 09 30 09 31 15 00 26 FF 7F
     75 10 95 02 81 02 09 38 15 81 25 7F 95 01 75 08
     81 06 95 01 75 08 81 01 C0 C0 05 0C 09 01 A1 01
     85 03 15 00 26 FF 0F 19 00 2A FF 0F 75 10 95 02
     81 00 C0 05 01 09 80 A1 01 85 04 15 01 25 03 95
     01 75 02 09 81 09 82 09 83 81 00 95 01 75 06 81
     01 C0 05 0D 09 01 A1 01 85 07 09 01 A1 00 05 0D
     09 30 15 00 25 FF 95 01 75 08 81 02 95 05 75 01
     09 32 09 42 09 45 09 44 09 3C 81 02 95 03 75 01
     81 01 05 01 09 30 09 31 15 00 26 FF 7F 75 10 95
     02 81 02 C0 C0

jalmeroth, 2024-11-19:
Does this issue persist or can we close this issue?

alankeny, 2024-11-19:
With the release of v0.66 this issue has been resolved.  We can close it.  Thanks!

jalmeroth, 2024-11-19:
Cool, thanks for confirming! You can close it yourself or @hrvach has to do it :wink:

closed 2024-11-19 by alankeny