hrvach/deskhop #151: Mouse pointer jump to parking position when change layer on QMK keyboards

nguyenhaiac, 2024-09-11:
I'm using  a keyboard with qmk firmware (vial-qmk fork).
Whenever I using a layer change key, they mouse pointer jump to the top right corner (parking position). Also this only happen in my Linux Machine.

Here is my setup: 
Linux (A device) - Windows (B device).
When the pointer is in the Linux machine, when pressing layer change, pointer jump to parking position
When the pointer is in the Window Machine, pressing the layer change key doesn't affect mouse position.

Not sure how it could happen since theoretically the layer change key doesn't send anything to the host device.

Please let me know if I can provide any further information.

hrvach, 2024-09-11:
Can you usbhid-dump -e all with keyboard connected directly and then use the layer change key to record what's getting sent?

nguyenhaiac, 2024-09-11:
Hi here is the usbhid-dump content

keyboard plugged via deskhop
` 001:005:001:DESCRIPTOR         1726064544.148617
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
 81 01 05 01 09 30 09 31 16 00 80 26 FF 7F 75 10
 95 02 81 06 09 38 15 81 25 7F 95 01 75 08 81 06
 95 01 75 08 81 01 C0 C0

001:005:000:DESCRIPTOR         1726064544.149475
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
 05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
 81 00 C0 05 01 09 02 A1 01 85 02 09 01 A1 00 05
 09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
 01 75 03 81 01 05 01 09 30 09 31 15 00 26 FF 7F
 75 10 95 02 81 02 09 38 15 81 25 7F 95 01 75 08
 81 06 95 01 75 08 81 01 C0 C0 05 0C 09 01 A1 01
 85 03 15 00 26 FF 0F 19 00 2A FF 0F 75 10 95 02
 81 00 C0 05 01 09 80 A1 01 85 04 15 01 25 03 95
 01 75 02 09 81 09 82 09 83 81 00 95 01 75 06 81
 01 C0 05 0D 09 01 A1 01 85 07 09 01 A1 00 05 0D
 09 30 15 00 25 FF 95 01 75 08 81 02 95 05 75 01
 09 32 09 42 09 45 09 44 09 3C 81 02 95 03 75 01
 81 01 05 01 09 30 09 31 15 00 26 FF 7F 75 10 95
 02 81 02 C0 C0

Starting dumping interrupt transfer stream
with 1 minute timeout.

001:005:000:STREAM             1726064546.436398
 02 00 FF 7F 00 00 00 00

001:005:000:STREAM             1726064546.556242
 02 00 FF 7F 00 00 00 00
`

when keyboard plugged dirrectly

`003:002:002:DESCRIPTOR         1726064671.404058
 05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
 81 01 05 01 09 30 09 31 15 81 25 7F 95 02 75 08
 81 06 09 38 15 81 25 7F 95 01 75 08 81 06 05 0C
 0A 38 02 15 81 25 7F 95 01 75 08 81 06 C0 C0 05
 01 09 80 A1 01 85 03 19 01 2A B7 00 15 01 26 B7
 00 95 01 75 10 81 00 C0 05 0C 09 01 A1 01 85 04
 19 01 2A A0 02 15 01 26 A0 02 95 01 75 10 81 00
 C0

003:002:001:DESCRIPTOR         1726064671.404459
 06 60 FF 09 61 A1 01 09 62 15 00 26 FF 00 95 20
 75 08 81 02 09 63 15 00 26 FF 00 95 20 75 08 91
 02 C0

003:002:000:DESCRIPTOR         1726064671.405266
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 05 07 19 00
 29 FF 15 00 26 FF 00 95 06 75 08 81 00 05 08 19
 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 C0

001:005:001:DESCRIPTOR         1726064671.406575
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
 81 01 05 01 09 30 09 31 16 00 80 26 FF 7F 75 10
 95 02 81 06 09 38 15 81 25 7F 95 01 75 08 81 06
 95 01 75 08 81 01 C0 C0

001:005:000:DESCRIPTOR         1726064671.407167
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
 05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
 81 00 C0 05 01 09 02 A1 01 85 02 09 01 A1 00 05
 09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
 01 75 03 81 01 05 01 09 30 09 31 15 00 26 FF 7F
 75 10 95 02 81 02 09 38 15 81 25 7F 95 01 75 08
 81 06 95 01 75 08 81 01 C0 C0 05 0C 09 01 A1 01
 85 03 15 00 26 FF 0F 19 00 2A FF 0F 75 10 95 02
 81 00 C0 05 01 09 80 A1 01 85 04 15 01 25 03 95
 01 75 02 09 81 09 82 09 83 81 00 95 01 75 06 81
 01 C0 05 0D 09 01 A1 01 85 07 09 01 A1 00 05 0D
 09 30 15 00 25 FF 95 01 75 08 81 02 95 05 75 01
 09 32 09 42 09 45 09 44 09 3C 81 02 95 03 75 01
 81 01 05 01 09 30 09 31 15 00 26 FF 7F 75 10 95
 02 81 02 C0 C0

Starting dumping interrupt transfer stream
with 1 minute timeout.

003:002:002:STREAM             1726064671.408843
 02 00 00 00 00 00

003:002:002:STREAM             1726064672.729216
 02 00 00 00 00 00

003:002:002:STREAM             1726064672.811177
 02 00 00 00 00 00
`

nguyenhaiac, 2024-09-15:
an update for this:

This issue actually happens on Windows as well, if I have only one Windows machine plugged to they deskhop, the pointer also moves to the parking position when pressing layer change.

I suspects that this happens in the primary machine (A device/only device)

heyuch, 2025-04-16:
@nguyenhaiac hi, I have encountered same issue, do you have any progress on this?

nguyenhaiac, 2025-04-16:
> [@nguyenhaiac](https://github.com/nguyenhaiac) hi, I have encountered same issue, do you have any progress on this?

unfortunately, no I haven't had any progress with this.