hrvach/deskhop #153: Not compatible with Adder XDIP KVM

spjen, 2024-09-18:
My setup is Adderlink XDIP (An IP KVM extender) for computer A and B.
I'm using an Apple Keyboard and Logitech trackball mouse, both of which work with DeskHop in my other setups.

When I try to use DeskHop with the XDIPs, both see it but they don't respond to inputs.

There is a device details page which lists the following for the plugged in DeskHop.
Speed(Mbps): 12
Protocol Version: 2.00
Maximum Power Consumption: 500mA
USB Type: usbhid

spjen, 2024-09-18:
Tested further. When in XDIP menus, the XDIP responds to inputs from the DeskHop but is not passing those along to the connected computers.

hrvach, 2024-09-19:
I honestly don't know what the problem might be here, I don't have this device to debug and it's very hard to guess remotely, especially when you mentioned device responds to inputs but not forwarding them on. I'll try shuffling report descriptors to make a boot mode keyboard "compatibility mode", maybe that will help.

spjen, 2024-09-19:
Thank you! I reached out to the manufacturer of XDIPs, and will update if I hear back.

qq8322302, 2024-10-26:
Additionally, for users who prioritize safety and do not want to use BLE, we can use wired USB. I also have a better idea that can be extended to hardware solutions with 4 or even 8 hosts. 
It can be expanded using RP2040 * 1+analog switch+CH9329 * N, and the price is also not expensive. 
RP2040 is connected to a keyboard and mouse, and at the same time, it is connected to 4 analog switches through UART * 1. By selecting which CH9329 to communicate with through GPIO, it can easily communicate with 1 out of N PCs. The disadvantage is that it can only communicate with one CH9329 at a time and cannot communicate simultaneously. 
If interested, I can also create this form of hardware, but I also need your software support.

[image]