hrvach/deskhop #154: Compatability with hid-remapper?

222Phoenix, 2024-09-20:
Is there a way to get deskhop to work with this?

https://github.com/jfedor2/hid-remapper/tree/master

hrvach, 2024-09-20:
Did you try it? I think there is a chance it might work, but I've never tried. Looks like a very cool project!

222Phoenix, 2024-09-20:
Yes, it doesn’t work. It’s hard to explain because it’s hard to tell what’s going on. Cursor movement rapidly presses all kinds of random keyboard keys.

222Phoenix, 2025-03-17:
Update, it's working with Desktop 0.71 at the moment. It may depend on one or both firmwares.

hrvach, 2025-03-18:
> Update, it's working with Desktop 0.70 at the moment. It may depend on one or both firmwares.

That's great to hear! Some bugs were indeed fixed for 0.7...