hrvach/deskhop #155: El Capitan mouse wont work

tgrrr, 2024-09-21:
Hi,

Big fan, love the project.

It works great with a wired mouse and keyboard. Everything works as expected.

When I try with El Capitan, the mouse wont move between monitors. It gets stuck on the non-El Capitan monitor.
- Keyboard works
- Mouse doesn't. It disappears from the first device as expected. It just doesn't appear on the second device

I've tested:

- Two devices with El Capitan installed (combined with a Mac or Windows machine)
- Multiple Mice, keyboards, etc.
- Everything in the FAQ
- The mouse works great when connected directly to the computer (without the deskhop)
- El Capitan can see the DeskHop Switch as a USB device

Both devices load El Capitan via SSD, is the only thing that's unique about the setup, and in common between the two devices both running El Capitan

El Capitan (10.11.6)
Deskhop (0.65)

Other devices include:
- Mac M1
- Windows 10

hrvach, 2024-09-21:
If you move the mouse towards the El Capitan box, other board led lights up and nothing happens? Can you try entering the "gaming mode" when the pico led is on on the El Capitan output to see if you can get mouse to work at all?

tgrrr, 2024-09-22:
> If you move the mouse towards the El Capitan box, other board led lights up and nothing happens?

Yes, that's spot on.

> Can you try entering the "gaming mode" when the pico led is on on the El Capitan output to see if you can get mouse to work at all?

Gaming mode fixes the problem, thanks. Happy to close this as resolved.

Out of interest:

One strange side-effect is _sometimes_ the mouse stops 3/4 of the way across the M1, when I connect to an external (wide) monitor. This only seems to happen while in gaming mode where the other computer is the El Capitan machine.

Wide monitor:
LG 34WL500
2560 x 1080 resolution

Unplugging the computer's USB connections and plugging them in again generally resolves the problem.