hrvach/deskhop #156: Mouse Dongle not registering

tgrrr, 2024-09-22:
Hi,

Extra excited about this project now my first issue is resolved.

I can't get any mouse dongles to work as per the FAQ.

> >    Will this work with keyboard/mouse combo dongles, like the Logitech Unifying receiver?

> It should work. After a recent FW update, support for combo receivers should be much better.

Source: [FAQ](https://github.com/hrvach/deskhop?tab=readme-ov-file#faq)

Tested with:

- Logi unifying receiver and [Logitech MX Vertical mouse](https://www.logitech.com/en-us/products/mice/mx-vertical-ergonomic-mouse.910-005447.htm) LED's flick as expected. Mouse works via USB cable and deskhop, not with dongle.
- [Kensington Expert Mouse Wireless Trackball M0186-M](https://www.kensington.com/en-au/p/products/control/trackballs/expert-mouse-wireless-trackball-1) LED's flick as expected.

- Tested on machines with and without Logitech and Kensington software installed.

Tested with:
- Deskhop 0.65
- Mac M1 Sonoma 14.6.1
- Mac El Capitan (10.11.6)
- Windows 10

Cheers

AsocPro, 2025-01-09:
Just as a data point on Logitech Dongles I have tested the unifying and bolt reciever on an Intel Mac on Sequoia 15.2 (and probably a few previous versions) and Fedora 41/OpenSUSE Tumbleweed systems It seems to be working for me with an MX Ergo as well as a MX Master 3. I just flashed to 0.65. Are you running pretty stock settings or have you customized a bit? Just trying to think of anything I have ran into that could help.

hrvach, 2025-01-10:
Same things in both ports? I have just one logitech dongle for the keyboard and that seems to work fine but I haven't tested the unifying receiver recently ... I'll try to borrow one if possible and test. Logitech gear is quite popular so this definitely needs to be fixed.