hrvach/deskhop #157: MacOS Keyboard A2520 via USB

tgrrr, 2024-09-22:
Hi,

Keyboard fun as well. Different issue, different ticket.

Will not work with this keyboard, connected via USB-A.

[MacOS Keyboard A2520 via USB](https://www.apple.com/au/shop/product/MK2C3ZA/A/magic-keyboard-with-touch-id-and-numeric-keypad-for-mac-models-with-apple-silicon-us-english-white-keys)

Tested with:
- Deskhop 0.65
- Mac M1 Sonoma 14.6.1
- Mac El Capitan (10.11.6)
- Windows 10

Cheers

steveh, 2025-10-31:
This model keyboard also doesn't work for me. Anything I can do to help debug?

I see two short flashes on A, a long flash on B, repeating. Cursor control switches between them.

hrvach, 2025-11-05:
Thank you for the dump, I think I figured out the issue. Will try to patch tomorrow, would you be able to test?

steveh, 2025-11-05:
Yep, sure thing

steveh, 2025-11-06:
Result of typing `qwerty`:

```
❯ sudo usbhid-dump --entity=stream -d 05ac:029f
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:026:000:STREAM             1762390568.006455
 90 03 63

001:026:000:STREAM             1762390568.014453
 90 05 64

001:026:001:STREAM             1762390569.534570
 01 00 00 14 00 00 00 00 00 00

001:026:001:STREAM             1762390569.614572
 01 00 00 00 00 00 00 00 00 00

001:026:001:STREAM             1762390569.870497
 01 00 00 1A 00 00 00 00 00 00

001:026:001:STREAM             1762390569.934569
 01 00 00 00 00 00 00 00 00 00

001:026:001:STREAM             1762390570.182514
 01 00 00 08 00 00 00 00 00 00

001:026:001:STREAM             1762390570.254504
 01 00 00 00 00 00 00 00 00 00

001:026:001:STREAM             1762390570.502544
 01 00 00 15 00 00 00 00 00 00

001:026:001:STREAM             1762390570.566592
 01 00 00 00 00 00 00 00 00 00

001:026:001:STREAM             1762390570.822582
 01 00 00 17 00 00 00 00 00 00

001:026:001:STREAM             1762390570.894645
 01 00 00 00 00 00 00 00 00 00

001:026:001:STREAM             1762390571.110598
 01 00 00 1C 00 00 00 00 00 00

001:026:001:STREAM             1762390571.182517
 01 00 00 00 00 00 00 00 00 00
```

steveh, 2025-11-07:
I flashed each side and no difference unfortunately!

hrvach, 2025-11-07:
That is odd, I tried emulating a keyboard with this descriptor and your messages provided, then plugged the new firmware to the computer and was able to get keystrokes recognized. I will try a few more things, but I'm not near my gear at the moment so it will be attempts "in the blind"

steveh, 2025-11-08:
Is that what the LED pattern is indicating? (two short flashes on A, a long flash on B, repeating) - that it can't recognise the keyboard? If it helps, it momentarily registers mouse movement on the flash on B. No keypresses registered.

I have a small amount of electronics knowledge so if attaching a raspberry pi debug probe or similar will help I'm happy to try if you point me in the right direction?

hrvach, 2025-11-08:
No, if the flashes are repeating, it means the device is reboot looping. Any chance your capacitor isn't properly soldered or the keyboard is drawing too much power? If you swap the keyboard and mouse ports, same behavior?

steveh, 2025-11-09:
I've actually tried two different (elecrow-preassembled) devices bought months apart that exhibit the same problem (I bought another to rule out a hardware fault). Other keyboards continue to work.

The keyboard draws up 160mA (0.8W) according to a power meter. If I plug it into the mouse side, just the keyboard side flashes twice, nothing on the mouse side.

hrvach, 2025-11-09:
I have an idea what to try next, but traveling for work so might be a bit slow. Does the constant flashing back-and-forth happen without you doing anything (e.g. pressing keys)

steveh, 2025-11-10:
Yes it does, plugged in either side it repeats without input.

hrvach, 2025-11-20:
That is super interesting. I tried emulating your descriptor and sending keys, and it worked with that patch, but it didn't crash the device into a reboot loop with either version. 

Can you plug the keyboard directly to a linux box and do: lsusb -v -d 05ac:029f ?

mglushko, 2026-09-05:
@tgrrr @steveh @hrvach I think the reboot loop is the same bug as #332, hiding on an interface the dumps never showed.

The keyboard has three HID interfaces. The third one, Touch ID, uses a bulk endpoint, so Linux never attaches its HID driver to it. usbhid-dump only shows the other two, and an emulator built from that dump only has two, which is why keystrokes worked in emulation and the loop never showed up. The lsusb -v output above does include the third interface's report descriptor, because lsusb prints it for an interface no driver has claimed. It declares three vendor reports of 107, 649 and 62 bytes.

DeskHop mounts that interface and parses that descriptor. The 649-byte report is the problem. The parser keeps one entry per field of a report in a table of 128 entries, this report has 649 fields, and the parser writes past the end of the table. The first thing it overwrites is its own pointer into that table, and the parse ends with a write through a bad pointer. That is the same failure as the Gameball in #332, and #361 fixes it. The LED pattern matches: the first two interfaces mount and blink, the third crashes the board, it reboots, and the keyboard is enumerated again.

I checked this with [deskhop-hidtests](https://github.com/mglushko/deskhop-hidtests), which runs the firmware's own parser on a PC against real descriptors, using the bytes from this thread. Current main crashes on the Touch ID descriptor. With #361 it parses and ignores it, which is the right thing for a vendor interface. The keystrokes from the stream capture decode correctly either way.

The media keys are a separate, smaller problem. They are on report ID 0x52, which is 82, and the table that routes reports only has room for IDs up to 23. #368 fixes that, the same way it fixes #367.

hrvach, 2026-09-22:
Release 0.80 should ideally fix this. Please reopen if it doesn't help and sorry for the inconvenience!

closed 2026-09-22 by hrvach