hrvach/deskhop #158: Certain keyboards lack LCtrl key - makes config access difficult

crushallhumans, 2024-09-23:
Hi - I use an Advantage2 Kinesis ergonomic keyboard in a MacOS config, and it lacks a LCtrl key. My control set is LCmd, LOpt, RCtrl, RCmd. When I need to reprogram the deskhop, I have to haul out a standard keyboard and swap it in, in order to trigger the LCtrl + RShift + C + O setting. Not a huge deal, but I figured I'd make you aware.

hrvach, 2024-09-23:
I definitely suck at choosing hotkeys. What's a reasonable key combo that all keyboards would have, but you would never trigger accidentally? Every single one I choose has a clash with some keyboard model :)

crushallhumans, 2024-09-23:
That does seem like a game of whack-a-mole. How long can the chord sequence be? Like can it be 'hold(either Ctrl + either shift) and then press in sequence (L, U, K, N, M)'? I've seen this multi-step command sequence used in IDEs, where huge numbers of hotkeys means a need for the second register of command keystrokes.

hrvach, 2024-09-23:
The key sequence is doable but you'd have to store past keypresses internally, and I feel very uneasy storing e.g. somebody's freshly typed password in a buffer. Yes, it's only internally, but zero memory (aside from output queue) is on purpose and I'd hate to break it. I'll ask LLM for some suggestions, maybe it is able to suggest something my tiny brain can't think of :)

crushallhumans, 2024-09-23:
That makes sense to me, and the rigor of never storing data at all is a good one and worth defending. Maybe it makes sense to just duplicate the invokation set, so (Lshift + Rctrl) || (Rshift + Lctrl) + C + O works - that solves for my issue, and I think it's a fair guess that all keebs will have an one set of opposed shift and ctrl. Unless they're fully programmable - which mine notionally is but I can't get into the utility to inject a new keycode into it :)  Thanks for the strong engagement - I love the product and hope it's successful!

hrvach, 2024-11-19:
There seems to be no shortcut I can think of that works consistently for everyone :-) 

What I will try to do is add several different shortcuts with hope at least one will be doable on everyone's keyboard. Would that work?

cornfeedhobo, 2024-11-19:
@hrvach fwiw, after looking at `keyboard.c`, maybe it's best to leave this concern for a little while - the code is easy enough to compile in a docker container. In the meantime, maybe better to update the docs on how to customize the shortcuts before compilation?

closed 2024-11-19 by hrvach