hrvach/deskhop #159: USB Drive "DESKHOP" does not appear /Volumes on macOS

adofse, 2024-09-29:
I just got Deskhop and trying to configure via Web configuration mode.

However, /Volumes/DESKHOP does not appear by pressing Left Ctrl + Right Shift + C + O.

Both micro USB are connected to two machines and keyboard side of LED is started blinking when I press  Left Ctrl + Right Shift + C + O.

Any way to mount Deskhop and use web configuration mode on macOS?

Firmware: pre-built v0.65.
Machine: MacOS Sonoma 14.6.1

hrvach, 2024-09-29:
Is the side blinking connected to MacOS?

adofse, 2024-09-29:
Yes.

hrvach, 2024-09-29:
I've tried it on Sonoma 14.6.1 and haven't seen any issues. Do you have any security software installed that would prevent usb drives from being mounted or a similar known restriction?

adofse, 2024-09-30:
Using different USB micro cable solved this issue!

closed 2024-09-30 by adofse