hrvach/deskhop #164: There is a better idea, Implementing HID KM with ESP32C3 BLE.

qq8322302, 2024-10-18:
Hello,hrvach:
    I am an electronics enthusiast from China and came across your project by chance. It is a fantastic project that can solve many operational problems with multiple hosts.
    I have an idea now, to use ESP32C3 BLE to simulate keyboard, mouse, and communication with multiple devices, and to add lithium batteries to get rid of wiring harnesses.

My thoughts:
    ESP32C3 serves as the main controller, and after turning on BLE, it broadcasts four names: KM1, KM2, KM3, and KM4. Four PCs are connected to each of these four BLE names.
    The IO of ESP32C3 can use software to simulate USB-HOST mode for connecting keyboards and mice.
(ESP32C3's ble_hid_device_demo: esp-idf-v5.2.1/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)
(ESP32C3 software simulation USB-HOST: https://github.com/sdima1357/esp32_usb_soft_host )
    ESP32C3-FH4， In China, it only costs RMB 5 yuan, and materials such as batteries and casings are also very cheap. I can design and produce PCB hardware, but my software level is not very good. I wonder if you are interested in researching together?We look forward to your reply.

[image]

chri2, 2024-10-20:
Sounds nice - I couldn't use it.

It will be a different device for end users without security needs.

My use case includes that I'm forbidden by rules at my work to use any keyboard that is connected wireless.

Making your project secure includes having a secure bluetooth stack which until now proved many times to be a wrong assumption.

Another point for me would be the needless use of batteries. That is not sustainable. There is no reason I can see that your idea couldn't work with one computer being connected through usb providing the power needed to drive your board (or just use an external usb power adaptor which in on most desks anyway.

I'd build or buy one for testing if it would work without a battery just for the fun of it and maybe to give it to my son to switch between his mobile and his desktop. On my desktop I couldn't use it for security reasons.

qq8322302, 2024-10-21:
@chri2    If for absolute safety, then wireless and lithium batteries are indeed not a good idea. Both will have certain security risks.
If it's for convenience and a clean desktop, then wireless and lithium batteries will definitely be more convenient.
This involves different usage scenarios.

tegger2000, 2024-10-23:
Maybe this project could be a base for your idea: https://github.com/Noskcaj19/hardware-kvm

hrvach, 2024-10-23:
This is a cool idea - I've investigated ESP32 initially but it didn't have usb. My plan is to expand the nunmber of devices one can control, my initial thoughts were most people who would use this need 2, a small number might need 3 and very  few people need 4 or more. I agree this would be a desirable feature for sure, I occasionally needed a third output, maybe something happens later this year :) 

Some of the technical limitations mentioned were actually chosen on purpose, like omitting Bluetooth - I simply don't believe there is a vendor who could implement this in a way that stands the test of time (https://github.com/engn33r/awesome-bluetooth-security for reference). Bluetooth stacks are often written by coders faced with unrealistic timelines and then never ever patched after vulnerabilities are discovered. The protocol itself is a bit too complicated for a secure implementation.

If you're still interested in helping out, I'd be curious how cheap this board could be made, i.e. what the ultimate price point is for small(er) batches. I'll be working on a new PCB variant when time permits...

qq8322302, 2024-10-26:
Firstly, let me share the desktop shop I created using RP2040 * 2. The PCB was redesigned and streamlined, matched with a metal casing, and the effect was very good.

[image]
[image]
[image]
[image]
[image]

----------------------------------------------------------

I found a similar project for ESP32C3 BLE: USB-HID-BLER, URL: https://github.com/dnstzzx/usb-hid-bler 
This project is to convert wired keyboard and mouse to BLE. I built a simple circuit and tested it. The mouse can be used normally, but the keyboard cannot be used. 
﻿  
I have already drawn the PCB according to my idea, and it will take about a week to produce it. I will test it first when the time comes. 
There are the following software issues that need to be addressed: 
1. I don't know how ESP32C3 can emit KM1, KM2, KM3, KM4, multiple BLE names. 
2. The compatibility of USB-HID-BLER is not perfect and needs to be addressed. 
3. Need to merge the functions of deshop into it. 

Regarding small batch costs: 
PCB+materials+shell+production, about RMB 55 yuan.  ($8) 

I don't know if you're interested in helping write software? I can provide you with hardware.

[image]

qq8322302, 2024-10-26:
Additionally, for users who prioritize safety and do not want to use BLE, we can use wired USB. I also have a better idea that can be extended to hardware solutions with 4 or even 8 hosts. 
It can be expanded using RP2040 * 1+analog switch+CH9329 * N, and the price is also not expensive. 
RP2040 is connected to a keyboard and mouse, and at the same time, it is connected to 4 analog switches through UART * 1. By selecting which CH9329 to communicate with through GPIO, it can easily communicate with 1 out of N PCs. The disadvantage is that it can only communicate with one CH9329 at a time and cannot communicate simultaneously. 
If interested, I can also create this form of hardware, but I also need your software support.

[image]

hrvach, 2024-11-06:
I really like your board and box! Unfortunately I don't have the time resources to rework it with ESP32, but if you want to work towards redesigning some of the existing hardware please let me know :)

qq8322302, 2024-11-06:
I have done some research on the ESP32C3 and wanted to implement desktop functionality, but it turned out to be quite difficult and challenging. Additionally, the security of BLE is not very high, so I decided to abandon the ESP32C3-BLE.
I have redesigned a new board as described above, using 1*RP2040 + 4*CH9329, which can control up to 4 devices. The hardware is already in production and should be available around the 8th.
This hardware architecture can easily control 4 devices or even more, with theoretical support for up to 254 devices. Additionally, the software code modifications for desktop functionality are minimal; it only requires replacing the USB devices section with UART.
Currently, I am trying to modify your software, but it's somewhat challenging for me. Would it be possible to invite you to collaborate on the software development?

[image]
The circuit board is already in production and should arrive in my hands around November 8th.

teuchezh, 2024-11-08:
In general, seamless switching of a Bluetooth keyboard and mouse between devices sounds cool

qq8322302, 2024-11-11:
[image]
I am trying to modify the deskhop software, but it will take some time.
PCB size: 94mm * 49mm

hrvach, 2024-11-11:
That's pretty cool, I like the board! Maybe consider adding digital isolators to the outputs?

What's the availability of those WCH9329 and what's the chip next to the left USB-A connector?

qq8322302, 2024-11-12:
1、 The CH9329 is currently in mass production and can be easily purchased, with a price of about $0.3-0.4$ (RMB: 2-3).
2、 The chip next to the left USB-A connector is a USB-HUB chip, which is used to expand the USB-HOST port of the RP2040.
In the Pico-PIO-USB project, I was unable to make it support two USB-HOST ports, so I used a USB-HUB chip for expansion.
3、 The two USB-A ports do not require an additional ESD chip, as ESD protection is already integrated into the USB-HUB chip.
4、 4*CH9329 USB buses might require an additional ESD chip, but I don’t think this is very important.
5、 I do not plan to add digital isolators because, after adding them, the four CH9329 Type-C ports would no longer be able to power the RP2040, which would require an additional power supply for the device. Once the software for the RP2040 Type-C port is debugged, I intend to remove it and not use it for power. 
  If digital isolators must be added, then four would be required to isolate all four CH9329 chips. Additionally, the RP2040’s Type-C port would need to be retained for power supply purposes.

frostworx, 2024-11-13:
I found this [bluetooth-usb-peripheral-relay project](https://github.com/bahaaador/bluetooth-usb-peripheral-relay) _(Bluetooth to USB HID relay using a Raspberry Pi Zero )_ earlier today on [ycombinator](https://news.ycombinator.com/item?id=42125863),

which reminded me that I recently wanted to check for a new deskhop firmware.

So here we are - maybe interesting for you :)

qq8322302, 2024-11-16:
[image]
Here’s the English translation:

I have modified some of the software functions on the original board, and now it is able to support the communication between 4 PCs. Due to issues with the USB hub chip and other factors, I redesigned the PCB, adding digital isolation chips to isolate the 4 PCs from the board. The board is powered via TYPE-C from the RP2040. The CH9329 has a maximum UART baud rate of 115200, so the theoretical maximum mouse report rate is around 400Hz. During testing, I found that when a 1kHz mouse was connected, the movement became sluggish. Optimization will be needed in the software for this in the future.The new PCB is expected to arrive in about 7 days, and I will conduct testing once it arrives.

hrvach, 2024-11-18:
Yeah 115200 is pretty sluggish, but I've ordered one sample chip to play around with it and test how feasible it would be. Interesting concept you've got here. I'm toying with the idea of putting a very low cost linux system-in-package like the f1c100s to handle all the incoming usb host work leveraging the linux kernel drivers for all the strange usb hardware out there and some super-cheap usb-capable microcontrollers for the outputs.

qq8322302, 2024-11-18:
[image]

Is it as I understand it from the image above? Using the Linux kernel driver to handle USB transactions is a great idea, as it allows for better compatibility with various devices. The MCU here would also need to develop its own software. If you need such hardware, I can also make it for you.

hrvach, 2024-11-19:
Yes, this is exactly what I had in mind (plus some usb isolators :-))) and I have to say I'm impressed how quickly and accurately you manage to iterate through designs. This would allow for just 1 master without the complicated logic to sync with the other node, and modular slaves in any quantity desired. I'm not sure how much more expensive it would get though, albeit f1c100s is quite affordable from what I've seen. For the USB MCU, something affordable like STM32F042 with internal flash and no requirements on external quartz oscillator. 

I'm waiting for one f1c100s board in the mail already, so then I'll try playing with it and see how quickly I can get it to boot and how reliable it would be. I think this would resolve any compatibility issues people might have and also allow for virtually unrestricted number of outputs.

qq8322302, 2024-11-19:
The F1C100S typically takes 3-5 seconds to boot and enter the system. (Set the boot wait time to 0, and try to keep the drivers as minimal as possible).
I suggest you first test with the F1C100S to check its compatibility with various unusual keyboard devices when used as a USB host.

**Below is a rough diagram I’ve drawn.** It uses STM32F042F6P6 * 4 as the MCU connecting 4 PCs, with ADUM1201BR*4 as isolators for serial communication. The F1C100S serves as the main controller, and the USB-OTG interface is connected to a USB-HUB chip, which provides 4 USB-HOST interfaces for connecting keyboards/mice. An additional Type-C interface is included for power supply, while also connecting to the CH340N chip for a serial bridge, which links to UART0 of the F1C100S for debugging.

Regarding configuration via a web page, I think the STM32F042's USB can virtualize CD-ROM/HID/keyboard/mouse at the same time, using the CD-ROM to store the web page and using HID for communication.

All UARTs of the STM32F042 can be connected to UART1 of the F1C100S, with different addresses used to distinguish them in the custom communication protocol.

Finally, there's one question: The F1C100S can only download firmware via USB-OTG, but currently the USB-OTG is occupied by the USB-HUB. How should we download the firmware? (Should we add an SD card and use it to boot the system? I don't think this is a good solution.)

The overall cost, including PCB + materials + metal enclosure + production cost = (approximately $15).

[image]
[image]

hrvach, 2024-11-19:
As for the boot time, I am secretly hoping to get it down to 1 or 1,5 seconds, otherwise it would be just too annoying. For the firmware, I'll see what's possible once I receive the chip - with the post office acting up these days, it might take a while.

I think TI ISO7721DR is cheaper than ADUM. One isolator could be removed and the soc connected to that output port side without isolation, maybe even powered by it. I wonder if there are more affordable SOCs like Nuvoton NUC980 series that have a bunch of usbs so one could remove a bunch of chips and maybe replace them with one usb isolator per port maybe. Or ditch everything and leave just one FPGA and synthesize a risc-v + 2 USB host controllers and a couple of devices :)

qq8322302, 2024-11-20:
The NUC980 is quite expensive at around $8; at this price, you could buy 11 STM32F042s. 
The ISO7721DR and ADUM are roughly the same price, mostly around $0.36 each, so four would cost about $1.44. 
The F1C100S is approximately $2.2, and a metal case is about $2. 
Four STM32F042s would cost around $3. 
The factory production cost per unit is $3 (for a small batch of 30-50 units, the more you produce, the cheaper it gets). 
The Type-C connector + resistors and capacitors + flash + USB hub + power IC + PCB is about $3.4... 

The isolation chips for the USB bus are much more expensive than the UART isolation chips (ISO7721DR / ADUM).
I'm not sure what your ideal price point is. Perhaps we can look into streamlining or removing some components to reduce costs. Could you share your target price or the budget constraints you're working within?

hrvach, 2024-11-27:
Well, my idea is to make it as cheap as theoretically possible without sacrificing safety or usability. I've also noticed nxp i.mx6ulz running around 4$ in quantities, and can boot sub-second. I'll see what f1c100s can pull off in terms of speed and power consumption. Do you know is there any documentation for f1c100s? I was only able to find https://linux-sunxi.org/F1C100s with a 21 page document. 

It would ideally fit into a 15$ budget altogether. Do you think that's doable? Many thanks for your kind help!

qq8322302, 2024-11-28:
The NXP i.MX6ULZ chip is quite difficult to purchase in China, and it comes with a very high price. Additionally, it requires extra components such as DDR memory and a power management chip, which further increase the overall cost. The BGA packaging also adds to the production costs.
A solution using the F1C100S + STM32F042 can be fully realized for around $14-15, which is much more affordable.
I'm also planning to buy an F1C100S for testing soon.
Here are some useful resources:
F1C100S Datasheet:  https://linux-sunxi.org/images/b/ba/F1C100s_Datasheet_V1.0.pdf
Setting up the Compilation Environment: https://wiki.sipeed.com/soft/Lichee/zh/Nano-Doc-Backup/build_sys/docker.html
Compilation Tutorial: https://blog.csdn.net/Naisu_kun/article/details/123142958

qq8322302, 2024-12-02:
F1C100S+xt25f128 (16MB NorFlash), prune uboot, prune kernel, modify partition size, currently the fastest startup speed takes 8 seconds.
Suniv uboot takes 1 second.
Uboot takes 1 second to start.
In uboot, it takes 3 seconds to load the kernel from flash into memory.
The kernel startup process takes approximately 2-3 seconds.
Below is the video I recorded.

https://github.com/user-attachments/assets/6ada1e0c-90c5-4bad-b82b-6aec64eed17d

hrvach, 2024-12-02:
Can't see the video, but I managed to trim it down quite a bit (no filesystem, runs custom init from ramfs)... it's a very nice little chip, I like it.

[image]

qq8322302, 2024-12-03:
That's great. SPL U-Boot is the first to boot, but I'm not sure if it's convenient to load USB-related drivers in U-Boot. I tried, and found that I couldn't enable the USB host. If USB-related tasks could be accomplished within U-Boot, then there would be no need to boot the Linux kernel, and this way, it indeed could start up and work within 1-2 seconds.

----
Normally, booting U-Boot and the Linux kernel involves reading the kernel from flash, which wastes a lot of time. The read speed is approximately 1MB/s.
[image]

hrvach, 2024-12-03:
Having a Linux kernel is desirable in this case because a lot of the specialty HID drivers, fixes and workarounds you basically get for free (as well as reliable HUB operation).

Faster boot can be achieved with some tricks (well, a LOT of them) - you have SPL loading the kernel directly as opposed to loading the main u-boot, that saves you ~ 0.2 sec. Slimming the kernel way down (since all you need is usb, serial and running a single binary from ramdisk) helps, also providing the loop calibration values explicitly (lpj) saves a bit of time, reducing serial output to quiet, skipping some generic code in SPL, doing MMC with 52 mhz, etc... the time spent in BROM you can't do much about, but the rest is lightning-fast and I'd say could be doable.

qq8322302, 2024-12-04:
If that's the case, using the F1C100S as the main controller is indeed sufficient.
Moving forward, I can proceed with preparing the related hardware. I plan to use a combination of F1C100S + STM32F070F6P6 + ISO7721DR.
The STM32F070F6P6 is slightly cheaper than the STM32F042.
For the F1C100S: it will be equipped with 8MB of flash (perhaps 4MB would be enough?), featuring a USB-OTG interface connected to a USB-HUB to extend 2-4 USB HOST ports (in the future, upgrades to the SOC might be possible via a USB flash drive?).
Regarding the STM32F070F6P6: four UART ports from the STM32 microcontroller will connect to four ISO7721DR digital isolation chips and then link up to the UART1 port of the F1C100S. We intend to differentiate communication addresses within the UART communication protocol used by the STM32 (I'm wondering if you have any other suggestions? For example, could the F1C100S emulate four UARTs using regular GPIO pins, each connecting to one of the four STM32 MCUs?).
Do you have any other recommendations?

Before that, I need you to provide an example firmware with very simple functionality. The USB-OTG interface is set to USB HOST mode and can be connected to a keyboard/mouse to print all HID reports to UART0.This makes it easier for me to test some hardware functions.

hrvach, 2024-12-04:
I think that would work. Maybe not dealing with flash completely and just putting a microSD card reader in there would not be a bad idea? A corrupted storage or even initial bootstrap is quite complicated with an SPI flash. MicroSD is something you can easily remove, plug into a PC and fix whatever needs fixing. Images would be read-only and run from RAM completely, so only initial boot would be doing any reading.

The chip has 3 UARTs internally, and leaving one for console/debug makes sense. I'd probably go with uart0 = console, uart1 = outputs 1 and 3, uart2 outputs 2 and 4. This way on a 2-port variant you can have completely dedicated bandwidth, and on a 4-port you can differentiate using addresses, yet never have neighbors on a same port. 

This needs some more testing and proof of concept - I'm not sure what the highest serial baudrate is and if driver behaves OK, hardware docs specify a divisor register but there are also open questions regarding DMA support, usb otg functionality etc... I'll try testing the usb and some serial experiments soon :)

qq8322302, 2024-12-06:
Currently, I am designing a hardware PCB.
Regarding the selection of a USB MCU, I am considering the following options:

1. STM32F042F6P6, with 16K ROM + 6K RAM, does not require an external crystal oscillator but needs a 5V to 3.3V LDO. The cost is approximately $0.7 (MCU + LDO).

2. STM32F070F6P6, with 32K ROM + 6K RAM, requires an external crystal oscillator and a 5V to 3.3V LDO. The cost is approximately $0.43 (MCU + LDO + crystal).

3. STC8H2K08U, with 8K ROM + 2.5K RAM, does not require an external crystal oscillator and operates on a 5V supply. It doesn't require a dedicated programmer as it can be programmed directly through its built-in USB interface. The cost is approximately $0.17.
I have tested the UART baud rate of the STC8H2K08U and found it can reach up to 3Mbps (theoretically it can reach 6Mbps, but my USB-to-TTL adapter does not support 6Mbps, so I haven't tested this yet).

Specifications: https://www.stcaimcu.com/data/download/Datasheet/STC8H-EN.pdf

Would you be willing to use the STC8H2K08U? This could save approximately $2 in costs. If you have any other better suggestions, please let me know.

qq8322302, 2024-12-06:
Complete the placement of electronic components. There is no wiring yet. Use STM32 MCU. 94 * 48mm. 
I would like to test later whether UART1 (PD3/PD4) and UART2 (PE8/PE7) can communicate. 
The F1C100S in my hand still cannot enable USB OTG, I need to wait for your image.
[image]

ak66666, 2024-12-06:
Wouldn't it be more convenient (from the cable management perspective) to put all the connectors on one side?
That would allow to squeeze the USB-C connectors closer.
I.e. move the USB-A connectors to the top left corner, the power/debug into the very top left corner position. Then the components for each computer-connecting USB-c could be stretched below them.

qq8322302, 2024-12-07:
> Wouldn't it be more convenient (from the cable management perspective) to put all the connectors on one side? That would allow to squeeze the USB-C connectors closer. I.e. move the USB-A connectors [...]

@ak66666  Thank you for your suggestion. I will carefully consider placing all cables on one side.

qq8322302, 2024-12-07:
@hrvach  Do you have any thoughts on the suggestion of placing all cables on one side? 
I tested UART1 (**PA3/PA4**) and UART2 (PE8/PE7) and they can communicate normally. 
I also tested the USB HOST and found that after inserting it into the USB hub, the MMC0 interface of the SD card will be removed. I wonder if there will be any negative effects? 
Keyboard/mouse/USB flash drive inserted through USB hub can be recognized normally.

Insert the log of USB HUB
```
[   34.552387] musb-hdrc musb-hdrc.1.auto: Babble
[   34.590161] usb 1-1: new high-speed USB device number 2 using musb-hdrc
[   34.810717] mmc0: card 0001 removed
[   35.082218] mmc0: host does not support reading read-only switch, assuming write-enable
[   35.092347] mmc0: Problem switching card into high-speed mode!
[   35.098476] mmc0: new SDHC card at address 0001
[   35.113623] mmcblk0: mmc0:0001 SZYL 7.44 GiB 
[   35.128521]  mmcblk0: p1 p2 p3
[   35.566636] hub 1-1:1.0: USB hub found
[   35.580966] hub 1-1:1.0: 4 ports detected
```

Log of inserting mouse on USB HUB

```
[   72.100107] usb 1-1.2: new full-speed USB device number 3 using musb-hdrc
[   72.249217] input: Logitech USB Receiver as /devices/platform/soc/1c13000.usb/musb-hdrc.1.auto/usb1/1-1/1-1.2/1-1.2:1.0/0003:046D:C53F.0001/input/input0
[   72.330894] hid-generic 0003:046D:C53F.0001: input: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-musb-hdrc.1.auto-1.2/input0
[   72.364822] input: Logitech USB Receiver Mouse as /devices/platform/soc/1c13000.usb/musb-hdrc.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C53F.0002/input/input1
[   72.382752] input: Logitech USB Receiver Consumer Control as /devices/platform/soc/1c13000.usb/musb-hdrc.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C53F.0002/input/input2
[   72.398162] musb-hdrc musb-hdrc.1.auto: int hwep alloc failed for 1x32
[   72.404853] input: failed to attach handler kbd to device input2, error: -16
[   72.416311] input: Logitech USB Receiver System Control as /devices/platform/soc/1c13000.usb/musb-hdrc.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C53F.0002/input/input3
[   72.431559] musb-hdrc musb-hdrc.1.auto: int hwep alloc failed for 1x32
[   72.438153] input: failed to attach handler kbd to device input3, error: -16
[   72.454251] hid-generic 0003:046D:C53F.0002: input,hiddev0: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-musb-hdrc.1.auto-1.2/input1
[   72.497283] hid-generic 0003:046D:C53F.0003: hiddev1: USB HID v1.11 Device [Logitech USB Receiver] on usb-musb-hdrc.1.auto-1.2/input2
[   72.893567] musb-hdrc musb-hdrc.1.auto: int hwep alloc failed for 1x32
[   72.903495] musb-hdrc musb-hdrc.1.auto: int hwep alloc failed for 1x32
[   72.913503] musb-hdrc musb-hdrc.1.auto: int hwep alloc failed for 1x32

remove
 [   87.617312] usb 1-1.2: USB disconnect, device number 3

```

Log of inserting a USB flash drive into a USB HUB

```
[  104.760106] usb 1-1.2: new high-speed USB device number 4 using musb-hdrc
[  104.899211] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[  104.921795] scsi host0: usb-storage 1-1.2:1.0
[  105.932627] scsi 0:0:0:0: Direct-Access     Generic  Flash Disk       8.07 PQ: 0 ANSI: 4
[  105.949434] sd 0:0:0:0: [sda] 8501888 512-byte logical blocks: (4.35 GB/4.05 GiB)
[  105.971595] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  105.991358] sd 0:0:0:0: [sda] Write Protect is off
[  106.004294] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  106.027290]  sda: sda1 sda2
[  106.041112] sd 0:0:0:0: [sda] Attached SCSI removable disk

remove
[  110.549456] usb 1-1.2: USB disconnect, device number 4
```

ak66666, 2024-12-07:
> @ak66666 Thank you for your suggestion. I will carefully consider placing all cables on one side.

No, thank you, I am looking forward for getting your device when it becomes available.
Now I use Deskhop with two computers.
With 4 ports I should be able to reach to the third one (instead of RDPing into it and suffering from the slow connection), and also the phone.

hrvach, 2024-12-08:
@qq8322302 I am waiting on a sample of WCH IC - CH32V203 that one is supposed to be both cheap and widely available there, I'll see if I can get one working to do anything meaningful.

As far as STC8H2K08U goes, never heard of that one before. Can you point me out to e.g. Aliexpress page where a sample could be ordered? I'm open to trying out anything. 

Regarding MMC/USB - as long as initial MMC boot works with USB devices still plugged in, it's fine. I have to test this situation locally, been busy with other bugfixes... My idea is to use the MMC only for a super fast initial boot, loading kernel + micro ramdisk with just a single tiny binary to RAM and execute from there.

qq8322302, 2024-12-08:
The price of CH32V203F6P6 is basically the same as that of STM32F070F6P6, and a 3.3V power supply still requires a 5V to 3.3V LDO. 
CH32V203F6P6/STM32F070F6P6/STM32F042F6P6 are pin to pin compatible and can be replaced with each other. 
Price: STC8H2K08U<CH32X033F8P6<CH32V203F6P6<STM32F070F6P6<STM32F042F6P6 
If I use CH32V203F6P6/STM32F070F6P6/STM32F042F6P6, I think my PCB has already been designed and I can try to make one for testing. 
STC8H2K08U and CH32X033F8P6 may be slightly cheaper, but they are not pin to pin compatible, so I don't think it's necessary to consider them.

qq8322302, 2024-12-08:
[image]
[image]
These are two different layouts of MCUs, with STC8H being simpler and using fewer components. I plan to try making STM32 first.

Bunny-Head, 2024-12-10:
I missed this entired thread and was getting more hyped the more I read, amazing! All the improvements could be applied to the actual model? Replace micro usb with usb C is definivetly an improvement (trying to get rid of all non-USBC/A usbs here at home :P).

ak66666, 2024-12-10:
> Replace micro usb with usb C is definivetly an improvement (trying to get rid of all non-USBC/A usbs here at home :P).
https://www.etsy.com/ca/listing/1763041320/deskhop-jr-a-keyboard-and-mouse-switch

qq8322302, 2024-12-15:
[image]
The PCBs for STC8H2K08U and CH32V203F6P6 versions have arrived, and I just soldered a PCB for STC8H2K08U. 
F1C100S can be successfully started through TF card. I still need some time to do some necessary tests. After the tests pass, I can mail the PCB to you.

hrvach, 2024-12-15:
This looks pretty awesome! Any chance to try out the 32v203 version, since tinyusb has support for that one? Would your email be 8322302@qq dot com?

qq8322302, 2024-12-16:
My email address is yaodaohao@126.com
The PCB of CH32V203F6P6 version still needs to wait for 3 days, (and several electronic components are still missing and being shipped).

qq8322302, 2024-12-17:
[images.zip](https://github.com/user-attachments/files/18160763/images.zip)
This is the licheepi image I compiled, but I am unable to send attachments via email, so I have to upload it here.
This SD card image is a bit strange, as it seems to lack a rootfs partition, which prevents access to the shell command line.
Follow the steps below:
1. Burn NOR image to NOR flash.
2. Burn the SD card image to the TF card.
3. Insert the TF card onto the licheepi.
4. When powered on, the device will automatically load uboot and kernel from the TF card, and then automatically load rootfs from the NOR flash.

qq8322302, 2024-12-17:
I am debugging CH32V203F6P6, which only has one UART and shares pins with USB.
This leads to the inability to use UART1 after turning on USB.
I have confirmed with WCH's FAE that this model cannot meet our needs and can only be replaced with another model.
I will try STM32F070F6P6 later to see if UART pin mapping is possible.
I have successfully debugged STC8H2K08U, and both USB and UART can be turned on and communicate simultaneously. UART can communicate at 3Mbps.

qq8322302, 2025-01-15:
hi
May I ask if your vacation has ended? I think it's time to mail FC100S to you.

AJolly, 2025-01-25:
Will the support external power for USB devices? 
Is this only going to support mouse and keyboard or can you use a USB hub to then support switching additional USB peripherals? (For example a USB microphone)

qq8322302, 2025-01-26:
> Will the support external power for USB devices?
> Is this only going to support mouse and keyboard or can you use a USB hub to then support switching additional USB peripherals? (For example a USB microphone)

@AJolly   only going to support mouse and keyboard

qq8322302, 2025-01-26:
[f1c100s.zip](https://github.com/user-attachments/files/18548745/f1c100s.zip)
@hrvach   This is all the information about the f1c100s experimental board.

hrvach, 2025-01-28:
Thank you, it's on its way!

zzisbeauty, 2025-04-26:
看呆了。。。。。

太强了👍

qq8322302, 2025-08-21:
Hi .

Recently, a chip has been discovered. It can directly support manual and automatic switching of 4 ports without any programming. Many products are already using it. Here is the introduction of the chip:

https://www.wch.cn/products/CH9374B.html