hrvach/deskhop #165: USB/HDMI Dongle not detected at boot if Deskhop plugged to it

OrpheeGT, 2024-10-22:
Hello,

I'm facing a little issue since the beginning.

On my corporate laptop, I'm using a USB/HDMI dongle :
https://www.amazon.fr/gp/product/B0949111Q7/

After system boot, on Windows Logon page, usually, the HDMI plugged screen raise up.

But if DeskHop is plugged to it, it stays of, like USB is not detected at all.
As soon as I unplug the DeskHop USB from the dongle, I hear the Windows USB plug soon, and HDMI screen wakes up.

I usually login to Windows, and I can then re-plug the DeskHop USB to the Dongle.
I could probably replug before Windows Login, but I usually do like said above.

I don't know why it happens. And I don't know how to retrive logs about it as Windows is not logged in yet...

hrvach, 2024-10-23:
It's quite difficult for me to debug this without having the actual device at hand, not sure what I could do here. I have a similar one made by Anker and it just works. 

I also tried the Logitech one (don't remember which port I plugged it in tbh) and it worked. I'll move ahead and update the wiki, but regarding the usb dock - I can only guess.

OrpheeGT, 2024-10-23:
I fully understand, I throwed the bottle in the sea but did not expect to much ^^

hrvach, 2024-10-23:
I'd really love to help because usually solving an issue means solving it for many people, but this one is super difficult for me to debug without being able to actually see / dump traffic / etc.

OrpheeGT, 2024-10-23:
How could I try to dump traffic ? as the USB hub seems to be "disconnected" ... ?

I may try to sniff/dump traffic if you tell me how I could do it... ?