hrvach/deskhop #172: I can't get Keyboards to work when directly connected via USB

cittris, 2024-11-18:
Hey there,

I took delivery of my Elecrow assembled Deskhop. I installed the latest firmware.
I've tried 3 keyboards via USB and none are recognised. However, one of the keyboards has the option for a wireless dongle which works and is how i have been using it.
When I connect any of the keyboards, the lights do not flash. When I connect the dongle it flashes as expected.

I've tried:
- Multiple Keyboards
- Different cables, all USB-A to USB-C
- ENFORCE_PORTS to 1

Keyboards:
Keychron Q1
Royal Kludge RK61
I forget the 3rd

Is there anything else I can do?

Cheers,

hrvach, 2024-11-18:
That is super odd, I've just tried a Keychron V3 and it works with no issues. Is it behaving the same if you plug the keyboard in the mouse port? Can you try a different microusb cable - maybe the power delivered to the keyboard is not sufficient for proper operation?

So, I'd suggest:
- plug the keyboard into the mouse port, is it the same?
- use a different micro usb cable to the computer

If the dongle works, it's obviously wired up correctly but something might be preventing it from operating normally.

cittris, 2024-11-23:
Thanks for the replies.
I connected the keyboard to the mouse port and still not picking up my input.
The LEDs on the keyboard illuminate but the A -> B flash doesn't happen.
If i put the wireless dongle in the mouse port, it works.

I'll get some more cables and a USB hub to test.

cittris, 2024-11-27:
An update, I have tried a dozen different cables in a variety of configs.
Keyboard illuminates but is not functional.
The A -> B -> A flash that indicates it's working doesn't occur.
I've also tried another Windows laptop to no avail.

Will have to source a USB hub to test with.

Any other thoughts?

hrvach, 2024-11-27:
I'm a bit puzzled, it's hard to debug things remotely - keychrons work for me in general without any issues. Try wiping config first (rshift + f12 + d) and then try setting force_kbd_boot_protocol in settings to 1. Does that do anything for you? I'm not super optimistic it would, you can try bumping     blink_led(&global_state); in usb.c to the top of tuh_hid_mount_cb and check if anything blinks when you plug in the keyboard - if not, then the callback never gets invoked for some reason. Does the mouse work in that port?

cittris, 2024-12-08:
Excuse the slow update.
I tested with a USB hub and that didn't work.

I wiped config and set force_kbd_boot to 1 and it didn't work. 

`void tuh_hid_mount_cb(uint8_t dev_addr, uint8_t instance, uint8_t const *desc_report, uint16_t desc_len) {
    uint8_t itf_protocol = tuh_hid_interface_protocol(dev_addr, instance);
    
    /* Flash local led to indicate a device was connected */
    blink_led(&global_state);
    
    if (dev_addr >= MAX_DEVICES || instance > MAX_INTERFACES)
        return;`

I believe i did that correctly. If i did, there was no blink.
Funnily enough, my mouse (VXE R1 Pro) was working when i initially raised this and now no longer works. 
There is a flash/blink but no input. 
I feel like i have to be doing something wrong and i really want this work but im just not sure what misstep im taking.