hrvach/deskhop #185: Support for Apple Mighty Mouse

jalmeroth, 2024-12-08:
Hi,

I discovered a very exotic HID descriptor in Apples Mighty Mouse:

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x04,        //   Usage Maximum (0x04)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x95, 0x04,        //   Report Count (4)
0x75, 0x01,        //   Report Size (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x04,        //   Report Size (4)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x32,        //     Usage (Z)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x04,        //     Report Count (4)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x05, 0xFF,        //   Usage Page (Reserved 0xFF)
0x09, 0xC0,        //   Usage (0xC0)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 64 bytes
```

The buttons are currently not recognized, as there is no `HID_USAGE_DESKTOP_POINTER` usage set.

So I would like to add the following to `extract_data` (hid_report.c) to cover this:
```cpp
        {.usage_page   = HID_USAGE_PAGE_BUTTON,
         .global_usage = HID_USAGE_DESKTOP_MOUSE,
         .usage        = 0x0, // none
         .handler      = handle_buttons,
         .receiver     = process_mouse_report,
         .dst          = &iface->mouse.buttons,
         .id           = &iface->mouse.report_id},
```

This way the buttons will start working.
<hr>

Next up: the wheel…

The device is sending the reports in boot protocol.

Therefore we end up in `extract_report_values` (mouse.c):

```cpp
    /* Interpret values depending on the current protocol used. */
    if (iface->protocol == HID_PROTOCOL_BOOT) {
        hid_mouse_report_t *mouse_report = (hid_mouse_report_t *)raw_report;

        values->move_x  = mouse_report->x;
        values->move_y  = mouse_report->y;
        values->wheel   = mouse_report->wheel;
        values->pan     = mouse_report->pan;
        values->buttons = mouse_report->buttons;
        return;
    }
```

As the casting into `hid_mouse_report_t` doesn't incl. `Usage (Z)` we will end up with the wrong values.

So I was thinking if we still need this clause or if we could go straight to extracting the values from the raw_report in the following lines, now that we have the hid_parser functionality in place.

So in my world the `extract_report_values` (mouse.c) function would look just like this:

```cpp
void extract_report_values(uint8_t *raw_report, device_t *state, mouse_values_t *values, hid_interface_t *iface) {
    /* If HID Report ID is used, the report is prefixed by the report ID so we have to move by 1 byte */
    if (iface->mouse.report_id)
        raw_report++;

    values->move_x  = get_report_value(raw_report, &iface->mouse.move_x);
    values->move_y  = get_report_value(raw_report, &iface->mouse.move_y);
    values->wheel   = get_report_value(raw_report, &iface->mouse.wheel);
    values->pan     = get_report_value(raw_report, &iface->mouse.pan);
    values->buttons = get_report_value(raw_report, &iface->mouse.buttons);
}
```

This would probably be a bit slower I guess, as we would have to parse the report instead of casting. Do you see any other problems with this approach? Let me know, what you think.

I would be happy to provide a PR for this, but this time I think starting the discussion upfront might make more sense. :wink: 

Greets,
Jan

jalmeroth, 2024-12-08:
Here's how a typical report looks like when it's parsed:
[image]

hrvach, 2024-12-08:
I think just removing:

```
         .usage        = HID_USAGE_DESKTOP_POINTER,
```

might be a better fix? Having two entries that point to the same handler and one is a wildcard accepting anything seems redundant. This raises a great point that "usage" here should not be mandatory anyway and it would probably improve device compatibility in similar cases. As always - great find :)

jalmeroth, 2024-12-09:
> This raises a great point that "usage" here should not be mandatory anyway […]

I don't know if you had in mind to remove it for all of them, but I started with the buttons only for now.

hrvach, 2024-12-09:
> I don't know if you had in mind to remove it for all of them, but I started with the buttons only for now.

Not all of them for now, I'm not sure if the rest are required as I only looked into the buttons one for now. I think it should work OK and improve compatibility across devices for sure, great work! I also need to create an action that auto-builds after a commit ... started doing some refactoring, hopefully the repo will look less crappy soon. :)

closed 2024-12-09 by hrvach

jalmeroth, 2024-12-09:
> I also need to create an action that auto-builds after a commit ...

Take a look here: https://github.com/jalmeroth/deskhopl/blob/main/.github/workflows/build.yml

I would be happy to assist with that!

PR #186 (2024-12-09): Allow buttons with and without pointer usage
Fixes #185