hrvach/deskhop #188: Screensaver will wakeup suspended device

jalmeroth, 2024-12-10:
Hi there,

here's another fun one :smile: When actively putting your maschine to sleep (suspend), it will get woken again once the screensaver is due to kick in.

I think we should add a clause to call `tud_ready()` before we actually queue the mouse report. What do you think?

Best regards,
Jan

hrvach, 2024-12-10:
Wouldn't that prevent you from waking up a computer if you wiggled the mouse on purpose? Maybe better put a tud_ready() check in the screensaver task among the initial checks?

hrvach, 2024-12-10:
Yes, that sounds good - maybe at the end of the checks to avoid calling it constantly if ss is not due to run yet? Would you be able to verify that it works as intended?

Man, you are catching so many of my bugs ... thank you!

jalmeroth, 2024-12-10:
```asm
   0x2000222c <+192>:	bl	0x200083f0 <tud_mounted>
   0x20002230 <+196>:	cmp	r0, #0
   0x20002232 <+198>:	beq.n	0x20002216 <screensaver_task+170>
   0x20002234 <+200>:	bl	0x20008400 <tud_suspended>
   0x20002238 <+204>:	cmp	r0, #0
   0x2000223a <+206>:	bne.n	0x20002216 <screensaver_task+170>
=> 0x2000223c <+208>:	lsls	r3, r7, #3
 ```

```
Thread 2 "rp2040.core1" received signal SIGINT, Interrupt.
0x2000223a in tud_ready () at /home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:86
86	  return tud_mounted() && !tud_suspended();
(gdb) finish
Run till exit from #0  0x2000223a in tud_ready () at /home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:86
core1_main () at /home/jan/workspace/deskhop/src/main.c:73
73	        for (int i = 0; i < NUM_TASKS; i++)
```

Jup, I think that works. :+1:

PR #189 (2024-12-10): fix: screensaver should not wake host from sleep
Fixes #188
closed 2024-12-10 by hrvach