hrvach/deskhop #191: External notification of switch state

AsocPro, 2024-12-18:
I assume this isn't something that isn't already integrated into the current iteration But based on how things are currently build would it be possible/difficult to mod in an external notifier of which of the different computers the input is being directed to?

What I am thinking is an i2c connection that could be fed into a qmk powered keyboard (which would have custom firmware to talk to the deskhop over i2c) and would be able to recieve input based on what computer the input was connected to. My biggest use case here is for using two different operating systems there are different key combinations that do the same thing like copy paste undo etc. On Linux control is used and on macOS command/super is used. This is kinda a mental annoyance that I have been working around and have actually modified my qmk layout to make easier.

I'm looking at building a new keyboard (ryanis.cool cosmos) and am looking to integrate in a mouse (probably the ploopy mouse as that can talk i2c pending a few software fixes) so I can do layer swaps for single hand typing from the mouse as well as other things. This made me realize it would be a perfect integration for the deskhop to be able to add some more logic to the mix and be able to enhance the overall experience of a dual computer setup.

This is a longer term project for me but I wanted to throw it out here to start checking things out and seeing what the different components I'm going to need to figure out to get this working. I'm happy to get my hands dirty and jump in but am just looking for a little guidance on where is the best place to get started with this.

AsocPro, 2024-12-18:
Yeah I had thought of something similar to that as well but figured i2c might be an overall cleaner approach.

I also just had another idea that could possibly work as well. I'm not sure if things work the way I understand but I'm pretty sure numlock can be activated/cleared on the computer/host side of things. Could it be possible to implement a numlock toggle on the deskhop side to use as a flag to denote which side things are connected to. That way it could be triggered back and forth and then used in logic on the keyboard side to send the different keypresses based on numlock presence. Also I totally get this would be a super custom setup but I'm just trying to come up with how this would be most optimally implemented.

hrvach, 2024-12-19:
This is already somewhat implemented - not with num lock but caps lock. It would be just a small patch to change that. You can select something that's rarely used, like scroll lock? Search for KBD_LED_AS_INDICATOR in the codebase and all should be revealed :) In usb.c you should be able to change KEYBOARD_LED_CAPSLOCK to KEYBOARD_LED_SCROLLOCK and change:

```
 leds = leds & 0xFD; /* 1111 1101 (Clear Caps Lock bit) */
```
to 0xFB mask (1111 1011) as scroll lock should be 0x02. Theoretically you could even leverage KEYBOARD_LED_COMPOSE or KEYBOARD_LED_KANA and use that in QMK.

AsocPro, 2024-12-19:
I love this project. I must have just not noticed/understood that setting when trying to look at things. That feels absolutely magical. 
I put the following in a macro and assigned it to a key in qmk:
```
      bool caps = host_keyboard_led_state().caps_lock;
      if (caps) {
        SEND_STRING(SS_TAP(X_A));
      }
      else {
        SEND_STRING(SS_TAP(X_B));
      }

```
And now it types 'a' on one computer and 'b' on the other. I'll need to do some playing to figure out the best way to actually implement the control/command switch as there are some nuances in how I want the layout to work but I think having the mapping on the qmk side is probably going to work better than mapping directly in deskhop itself because there are cases where I would want to use both the super+letter and command+letter on linux as well so I would want them to just be available with different keypresses which would be basically impossible to figure out which was meant from the deskhop side of things. I don't mind which led is used either I was just brainstorming so capslock is totally fine.

@hrvach You are an absolute champion.

Closing as this is already implemted.

closed 2024-12-19 by AsocPro

hrvach, 2024-12-19:
> And now it types 'a' on one computer and 'b' on the other. I'll need to do some playing to figure out the best way to actually implement the control/command switch as there are some nuances in how I want the layout to work but I think 

I would encourage you to share your setup / solution after it's done, as I'm sure there are more people who would find this very handy and convenient. Additional points would go to any solution that includes non-patched QMK, that is - enable anyone with a qmk/via/vial keyboard to define a different layer that would map to an OS on the currently active output.

Thank you for coming up with new and interesting ideas to further enhance usability!

AsocPro, 2024-12-19:
I'll for sure write something up after I figure out the best way to implement things. I have a few different ideas and need to get back up to speed on the QMK side of things because it has been a while since I have dug into the inner workings of things.