hrvach/deskhop #195: Feature Proposal: Support for Multiple Configurations

fabien88fr14, 2024-12-30:
Hi,

First, I want to thank you for the amazing work you’ve done. This device is truly a game-changer for me.

My current setup is as follows:

- Mac OS with two screens on the left
- Windows with one screen on the right

Sometimes, I prefer to swap the configuration. For instance, when I have a lot of work to do on my Windows workstation, I switch to:

- One screen on Mac OS
- Two screens on Windows

It would be fantastic to have a feature that allows users to switch between configurations using a keyboard shortcut.

Let me know your thoughts on this!

hrvach, 2025-01-08:
This is doable and should not even be too hard to implement, but I'm stretched very thin resource-wise so trying to prioritize fixing bugs first and improving compatibility. Please be so kind to add this request here:

https://github.com/hrvach/deskhop/discussions/27

and I'll try my best.

fabien88fr14, 2025-01-08:
Thanks a lot @hrvach. Appreciate your hard work.
I'll add it to the thread you mentioned.

yeathn, 2025-09-15:
+1 would also love this feature