hrvach/deskhop #205: Custom shortcuts please

ToxMox, 2025-01-12:
First, I really love my deskhop device! It's pretty sweet. 

I'd love the ability to specify custom keyboard (or mouse) shortcuts to control deskhop. Maybe that could be added to the web config someday?

For example I can map buttons on my mouse to whatever I want. I have a button on my mouse bound to F20. I'd love the ability to make F20 be my shortcut for switching desktops instead of Left Ctrl-CAPS.

That being said. In the meantime, if I want to modify the code to manually replace that bind to be F20 where would I find it? I'm not a programmer but I can probably figure out building from code etc.

hrvach, 2025-01-14:
Check keyboard.c, it's at the very beginning - remove the modifier and define HOTKEY_TOGGLE to HID_KEY_F20 in include/user_config.h I believe...

It's on the to-do list, but work is very busy so I'm trying to bugfix first to have most things working for most people and then move ahead with additional features. Apologies for being slow!

ToxMox, 2025-01-14:
Please don't worry about being slow 😀

{.modifier       = KEYBOARD_MODIFIER_LEFTCTRL,
     .keys           = {HOTKEY_TOGGLE},
     .key_count      = 1,
     .pass_to_os     = false,
     .action_handler = &output_toggle_hotkey_handler},
 
What's the proper way to remove the modifier? Do I set it to false or am I supposed to delete the line completely or something else? 

Thank you for your help!

ToxMox, 2025-01-14:
Ok so I successfully removed the modifier and left caps lock for now. I bound CAPS LOCK to a button on my mouse and saved it to the mouse memory. I can now switch desktops using just CAPS LOCK on the keyboard but the button I bound on the mouse doesn't trigger a switch and just passes it through. If I swap the keyboard and mouse ports then the bound button on the mouse does switch desktops but the keyboard no longer will.

hrvach, 2025-01-16:
Silly question (I forget stuff like this all the time) - you upgraded both boards with your recompiled firmware?

ToxMox, 2025-01-16:
> Silly question (I forget stuff like this all the time) - you upgraded both boards with your recompiled firmware? It should have upgraded the other one automatically if version number was bumped, but that's easy to forget.

No. I flashed using Config mode and the drive that mounts in PC 1. I thought I only had to flash on both computers if holding down the button while replugging the usb.

ToxMox, 2025-01-16:
Ok I manually flashed the other board and now it works! Thank you so much :)

closed 2025-01-16 by ToxMox

hrvach, 2025-01-16:
Normally you can upgrade just one side, but that requires a higher firmware version number than the one you are running... so happy you got it working! :)