hrvach/deskhop #207: Touchpad support

rnewman, 2025-01-14:
It appears that the DeskHop always reports the connected device as a mouse. On macOS, that means a USB-connected Magic Trackpad loses tap-to-click and two-finger scrolling. These work correctly through common KVM switches (e.g., TESmart).

Is it possible to:

1. Get this working? This feature alone is the difference between the perfect device and unusable for trackpad users.
2. Switch into a mode whereby switching between devices is entirely driven by keyboard shortcuts, much like a traditional KVM, and a non-keyboard HID device is passed through?

hrvach, 2025-01-14:
I'm not even sure how the HID report looks for the magic trackpad, can you share the descriptor please? For the classic switching you can use ctrl + caps lock and try the "gaming mode" which flips to relative mouse, but I'm not sure what that will do to scrolling and clicking.

It makes sense to try and implement support for the touchpad since there are many people who swear by it, but I'll have a better idea of scoping the problem once I know how the descriptor looks (hopefully). Thanks!

rnewman, 2025-01-14:
Raw dump from `mac-hid-dump`:

```
05AC 0265: Apple Inc. - Magic Trackpad
DESCRIPTOR:
  06  00  ff  09  0d  a1  01  06  00  ff  09  0d  15  00  26  ff 
  00  75  08  85  3f  96  0f  00  81  02  09  0d  85  53  96  3f 
  00  91  02  c0  
  (36 bytes)

05AC 0265: Apple Inc. - Magic Trackpad
DESCRIPTOR:
  05  01  09  02  a1  01  09  01  a1  00  05  09  19  01  29  03 
  15  00  25  01  85  02  95  03  75  01  81  02  95  01  75  05 
  81  01  05  01  09  30  09  31  15  81  25  7f  75  08  95  02 
  81  06  95  04  75  08  81  01  c0  c0  05  0d  09  05  a1  01 
  06  00  ff  09  0c  15  00  26  ff  00  75  08  95  10  85  3f 
  81  22  c0  06  00  ff  09  0c  a1  01  06  00  ff  09  0c  15 
  00  26  ff  00  85  44  75  08  96  6b  05  81  00  c0  
  (110 bytes)

05AC 0265: Apple Inc. - Magic Trackpad
DESCRIPTOR:
  06  00  ff  09  0b  a1  01  06  00  ff  09  0b  15  00  26  ff 
  00  75  08  96  04  00  85  e0  81  22  09  0b  96  01  00  85 
  9a  81  22  c0  06  00  ff  09  14  a1  01  85  90  05  84  75 
  01  95  03  15  00  25  01  09  61  05  85  09  44  09  46  81 
  02  95  05  81  01  75  08  95  01  15  00  26  ff  00  09  65 
  81  02  c0  
  (83 bytes)

05AC 0265: Apple Inc. - Magic Trackpad
DESCRIPTOR:
  06  00  ff  09  03  a1  01  06  00  ff  09  03  15  00  26  ff 
  00  85  c0  96  6b  00  75  08  81  02  c0  
  (27 bytes)
```

Let me know if you need anything else!

hrvach, 2025-01-16:
There is a chance

Usage Page (Digitizer),             ; Digitizer (0Dh)
Usage (Touchpad),                   ; Touch pad (05h, application collection)
Collection (Application),
    Usage Page (FF00h),             ; FF00h, vendor-defined
    Usage (0Ch),
    Logical Minimum (0),
    Logical Maximum (255),
    Report Size (8),
    Report Count (16),
    Report ID (63),
    Input (Variable, No Preferred),
End Collection,

this might be enough, but I'm unsure without doing some testing.

Other stuff is proprietary like:

Usage Page (FF00h),                 ; FF00h, vendor-defined
Usage (0Bh),
Collection (Application),
    Usage Page (FF00h),             ; FF00h, vendor-defined
    Usage (0Bh),
    Logical Minimum (0),
    Logical Maximum (255),
    Report Size (8),
    Report Count (4),
    Report ID (224),
    Input (Variable, No Preferred),
    Usage (0Bh),
    Report Count (1),
    Report ID (154),
    Input (Variable, No Preferred),
End Collection,

without obvious and apparent usage. Recording the USB traffic with and without multitouch might be helpful (as little traffic as possible to make it easier figuring out how that is sent).

In theory, it should be doable but I'll have to get the device first.

jakehu, 2026-03-18:
I am currently using a USB-A to Lightning cable to connect the trackpad to the DeskHop hardware. Both of my computers are Macs. I’ve found that while the cursor can move between the two Macs, neither of them supports gestures—only basic cursor movement works. I hope gesture support can be made compatible soon.

tthayer, 2026-11-09:
@hrvach Hello! Thank you so much for making this project. The simplicity is really remarkable. I have a branch with refined macos behavior and working multitouch support for the Apple Magic Trackpad 2. I'm not sure what the contribution guidelines are for this project, especially around AI-generated contributions, but if you are open to it I would like to create a PR for this.