hrvach/deskhop #211: Keychron Keyboard USB Dongle Not Working

apastuszak, 2025-01-20:
This is more a PSA that a bug. I have a Keychron V1 Max, that does wired, 2.4 Ghz wireless dongle, and Bluetooth. The USB wireless dongle does not work with the Deskhop. I think maybe the USB port on the Deskhop doesn't provide enough power to the dongle work.

Another keyboard I have, the Nuphy Air75 V2 works just fine wit the USB dongle, so it's definitely a Keychron issue and not a USB dongle issue.

hrvach, 2025-02-07:
New release with some potentially relevant bugfixes will be out soon, please be so kind to try that one and let me know if anything changes.

hrvach, 2025-02-09:
Please try v0.71 pre-release, does the issue still persist?

apastuszak, 2025-02-10:
I'll try tonight.

hrvach, 2025-02-25:
That's odd, I think I really need to work on a debug image that prints some information you could share... lacking the actual hardware, I can only guess what might be wrong. I'll try to work on this when I'm back from the trip.

geekbozu, 2025-03-14:
I also have issues with my Keychron, If you need more data points I can offer to take them!

apastuszak, 2025-03-14:
I'm going to plug the dongle into a powered hub and plug that into the deskhop and see if that makes any difference.

sk33ve, 2025-05-08:
Just a +1 on this - I just set this up today and it works *amazingly* but I cannot get the keychron dongle to work.
@apastuszak  - did your fix of using a powered hub help at all? I tried it but it seemed to make no difference for me.

sk33ve, 2025-05-30:
Happy to dump the descriptors if you can point me at how to do that :-) 

I’ll try and work out the cheapest option, but happy to act as debugger to save you the cash.  
I’m a software engineer by trade so relatively competent (despite not having a clue what you mean by dumping descriptors!)

hrvach, 2025-05-31:
Can you try https://github.com/todbot/win-hid-dump so we have the descriptor itself?

geekbozu, 2025-05-31:
Gladly. 

[USBC.Txt](https://github.com/user-attachments/files/20534250/USBC.Txt)
[USBA.txt](https://github.com/user-attachments/files/20534249/USBA.txt)

hrvach, 2025-06-03:
I have a theory what might be happening here. This seems fishy in the descriptor:

0x81, 0x03, /*      Input (Constant, Variable),     */

and the code basically checks if constant bit is set, then disregards that data since it considers it a constant.

Maybe try removing lines 86 and 87 in hid_report.c just for the sake of testing. Let me know if this helps?

geekbozu, 2025-06-03:
Still no dice with that change. If there is some form of debugging or other in code evaluation you think would make sense I can solder a debug header up to my spare deskhop at somepoint in the near future. Just don't want to go digging around blind. 
I can also take USBPCap/Wireshark dumps of both the USBA dongle directly connected and through deskhop

hrvach, 2025-06-04:
Wireshark would help, or even better, try usbhid-dump -e all and then type a few characters (ofc the dongle should be connected to the pc directly) on a Linux box. The Windows tool suggested above apparently has some limitations and might have reported the actual descriptor inaccurately - "The HID Report Descriptors displayed are reconstructed by the HIDSharp library from Windows HID data, and do not represent the exact descriptor sent by the device."

Any chance you have a Linux box/vm lying around? If not, Wireshark dump might also help... Thanks!

sk33ve, 2025-06-04:
I tried using WSL2 but it says No Matching HID interfaces. Maybe it is virtualised through WSL2?

If no one else has a handy Linux box, I can probably find an old Raspberry Pi I could use.

hrvach, 2025-06-06:
I think I might know what's going on here - single interface having multiple keyboards defined behind different report IDs, so the second one gets parsed last but reports coming with the first ID. This is a bit sucky handling and I'll have to look into fixing this.

Can you try to confirm by enabling NKRO for your keyboard and trying to use the dongle then? (maybe Fn + N if it's not qmk/via)

hrvach, 2025-06-08:
You can try verifying by doing usbhid-dump again and typing on the keyboard with the dongle plugged in your raspberry pi. If you see something like:

001:004:002:STREAM             1749067853.239361
 01 00 00 11 00 00 07 00 00

you are using report ID 01 which is probably the normal mode, but if you see 0C at the beginning of the packet, it's using ID 12 which should be the NKRO one.

hrvach, 2025-06-08:
Maybe try opening the manual to see how it could be toggled? I'm moderately certain this is the issue so I'll try to find some time to refactor this. It's not a oneliner fix sadly... i totally wasn't expecting this behaviour

sk33ve, 2025-06-09:
I've had a good look through the docs, and I believe it *should* be working.  Bit wired I consistently get 6KRO (using the Microsoft Ghosting Demo page), and wireless I get NKRO.  The toggle button doesn't seem to do anything in either case.  
I will submit a support request to them to assist, and report back.

sk33ve, 2025-06-12:
Support pointed me at the Keychron Discord (which I didn't even know existed!) And it looks like either there is something unusual about my keyboard, or not all keyboards support the NKRO toggle.  
Mine doesn't have it mapped to FN-N (I did a factory reset and it still isn't there), and if I do map it, it doesn't seem to function.
*so* it might be worth someone else trying it out to see if theirs works.  I've written back to support to ask for help as to why it is not working.  
I assume the fix you proposed @hrvach wouldn't require NKRO to be working, it is just a potentially useful workaround in the meantime?

geekbozu, 2025-06-12:
Ill give this a try on mine a little later!

justMaku, 2025-06-13:
I'm seeing this issue with another KVM (level1techs) and after going down this rabbit hole I can confirm that:
1. The Keychron Q-series seem to not implement the NKRO toggle properly, assigning this in the VIA does absolutely nothing no matter how many times do you press it.
2. Compiling my own QMK firmware based on the source code provided by Keychron with NKRO completely disabled does put the keyboard in the 6KRO mode when connected over USB, but the dongle itself still reports as NKRO.
3. I believe the issue is due to the gamepad/mouse reports/descriptors present in the dongle. I have another keyboard that uses a very similar dongle (exactly the same industrial design) that works just fine with my KVM over 2.4Ghz.

geekbozu, 2025-06-24:
https://github.com/geekbozu/deskhop/actions/runs/15839631538 Builds for anyone else we as well :) 

So I have a Keychron V5 Max. 
Main/Current Release. 
Normal: No Input
NKRO: No input

With the new requested change. 
Normal: No Input
NKRO: I typed this comment with it! Seems to be working but odd that NKRO has to be toggled. 

And no worries on the real life. It happens. Thanks for the awesome piece of gear.

geekbozu, 2025-06-25:
Good eye!
That was from a prior test this issue thread where we tried to allow constant hid reports through to see if it was just the keyboard doing something weird.. It did not work so I reverted it and added in the new requested change.

linickx, 2025-07-11:
Keychron K3 Max user here, reporting similar weirdness.

My _Host A_ is MacOS:

- With the 2Ghz dongle only the osx special keys like expose, brightness, volume work. (_Cannot type anything_)
- All keys work as expected when connected via USB cable.

Latest firmware on keyboard, updated with launcher.keychron.com. I'm not sure what NKRO is, didn't see anything alluding to that in the settings on the launcher site.

I'm happy to provide any debugs/outputs/etc if it helps.

hrvach, 2025-07-12:
Can you guys please test this one and tell me if it behaves the same?

[deskhop-test.zip](https://github.com/user-attachments/files/21199312/deskhop-test.zip)

sk33ve, 2025-07-20:
Yes, sadly the same for me too - made no difference - I can still use the media keys, but not the regular ones.  

Happy to try out anything else!

hrvach, 2025-08-16:
[multi-report-id-test.zip](https://github.com/user-attachments/files/21811502/multi-report-id-test.zip)

Can you try this one? I have no easy way to test this issue, so I'm afraid you'll have to be the test subject :) 

Sorry for the slow progress, work has been insane and it's very hard to find any spare time.

Let me know if it fixes anything.

cc: @linickx @sk33ve

linickx, 2025-08-17:
Working for me!  Using it now to type this response. Thank you!

hrvach, 2025-08-17:
Excellent, there should be a release coming soon with a couple of fixes.

sk33ve, 2025-08-17:
I've been away this last week but will try tomorrow. Thanks so much for this!

hrvach, 2025-08-17:
https://github.com/hrvach/deskhop/releases/tag/v0.72

Please test-drive it a bit, if it's OK we'll bump it to stable.

cc:
@geekbozu 
@apastuszak

sk33ve, 2025-09-06:
Just thought I'd report back after a few weeks of testing I've had absolutely no issues with this at all!  Briefly thought I did but it turned out to be a dodgy key on my keyboard (replacement keys enroute as we speak!). 

Thanks again for the amazing project and support @hrvach  - I just set up an MSF regular donation as a direct result of this so have double the good-karma! 

If I can help with any other aspect of it - please do give me a shout.

hrvach, 2025-09-13:
Thank you! Great to hear it's working.