hrvach/deskhop #212: distinguish between two deskhop on one computer

chri2, 2025-01-23:
I have two desks I'm working on, each of the desks having its own deskhop.

Keyboard and mouse are different in both places and I'd like to setup my OS to use different settings for each of the keyboard/mouse on the different two deskhops.

The deskhop connected to my computer looks in both places the same - I can't distinguish between deskhop1 (keyboard a and mouse b) and deskhop2 (keyboard c and mouse d).

Therefore I can't e.g. (re-)map keys on the OS-side for each of the keyboards differently.

A possible solution would be to allow to setup a deskhop with a different from the default usb id or maybe allow setting of a serial number that on other devices can be recognized by e.g. udevd.

AJolly, 2025-01-25:
You may be able to track based on which USB port you are connected to,?

When you say two desks are both desks being run from the same computer?

closed 2025-01-27 by hrvach

hrvach, 2025-01-28:
It's not actually set in the .h file (just the placeholder) but uses the unique ID gathered from the SPI flash which is the standard way these boards determine serial number. Each board should have an unique ID and ideally you'll be able to differentiate based on that.

chri2, 2025-01-29:
Thanks for clarifying!

chri2, 2025-01-30:
Just tested and it works. I had to struggle a bit with **udev on linux**. Here's my solution (Debian Bookworm):

#### Make udev ask the hardware database using deskhops serial number

On my system the default udev rules do not include the information about the serial number when asking the hardware database for extra information about a device.

`/etc/udev/rules.d/59-evdev.rules`:
```
ACTION=="remove", GOTO="evdev_end"
KERNEL!="event*", GOTO="evdev_end"

# Device matching the input device name and the S/N (used for Deskhop)
KERNELS=="input*", \
  IMPORT{builtin}="hwdb 'evdev:sn:$attr{name}:$attr{uniq}'", \
  ENV{.HAVE_HWDB_PROPERTIES}="1"

LABEL="evdev_end"
```

The filename is chosen, so that this rule would run before my default rule for evdev named `/lib/udev.d/rules/60-evdev.rules`. You might need to change the filename for your system accordingly. `/lib/udev.d/rules/60-evdev.rules`  contains `IMPORT{builtin}="keyboard"` which should run after the new rule.

#### Add extra information for the keyboard attached to deskhop

The device match above will lookup the hardware database for a section `evdev:sn:<deskhop name>:<deskhop s/n>`. I added it to the database:

`/etc/udev/hwdb.d/59-keyboard.hwdb`:
```
# deskhop with S/N
evdev:sn:Hrvoje Cavrak DeskHop Switch Keyboard:xxxxxxxxxxxxxxxxx
 KEYBOARD_KEY_7004a=insert
 KEYBOARD_KEY_70049=home
```

To get the `xxxxxxxxxxxxxxxxx` part (serial number) I used `lsusb -d2e8a:107c -v | grep -i serial` which outputs something like

```
# lsusb -d2e8a:107c -v | grep -i serial
  iSerial                 3 xxxxxxxxxxxxxxxxx
```
The first digit seperated by space from the S/N is not part of the number - in my example the `3`.

hrvach, 2025-01-30:
That's great, since you were kind enough to write this down, care to put it in a Wiki format and I'll add it to the project docs?

chri2, 2025-01-31:
I believe the wiki will accept md? The comment has already been written in md, but I guess it's only possible to extract those via the api in md again. If it could help anybody, please go ahead and put it onto the wiki! 

Thanks for the great project!

[comment.md](https://github.com/user-attachments/files/18617738/comment.md)

hrvach, 2025-02-02:
> I believe the wiki will accept md? The comment has already been written in md, but I guess it's only possible to extract those via the api in md again. If it could help anybody, please go ahead and [...]

Added, thank you again for being helpful and sharing!

commit cf842f82 (2025-01-27): Add support for device serial numbers.
This should resolve #212