hrvach/deskhop #213: MultiControl  -  possible clone?

AJolly, 2025-01-24:
Unsure if they are just using your source code

https://www.kickstarter.com/projects/electronics-maker/multicontrol-control-multiple-pcs-with-one-keyboard-mouse/

hrvach, 2025-01-24:
Hard to say, looks like 2 x rp2040s due to qfn56 form factor. I can only see one spi flash, the other is maybe on the other side or they did something like swd bootstrapping the second one.

Clones are totally fine as everything is open source, but they should still respect the GPL and publish the source if based on it. Attribution is also a fair thing to do :). Wishing them all the best with the project and hopefully it helps many more people get a better daily experience behind a computer.

AJolly, 2025-01-25:
Agreed, just hope they post source if it is a fork

eltariel, 2025-02-10:
> Hard to say, looks like 2 x rp2040s due to qfn56 form factor. I can only see one spi flash, the other is maybe on the other side or they did something like swd bootstrapping the second one.

If it's a clone, the SOIC-8 in the middle of the board is more likely to be the isolator than an SPI flash chip - it looks like there's a gap in the copper layer underneath it in one of the pics. I can't see anything else  on the top of the board that might be flash, so it's probably on the bottom.

I'm tempted to sign up for it anyway just to see 🤣

ak66666, 2025-02-10:
I asked and answer was:
> No, MultiControl does not use any part of the DeskHop project, including its code, schematics, or design. MultiControl is an independently developed product with a fully custom hardware and firmware solution designed from the ground up.