hrvach/deskhop #214: Mouse Extra Button Interuption

ahshift, 2025-01-25:
As usual, this project has been a god send finding i apprecaite all you've done working on this.

Im not sure if it this would be a feature request or Issue.

When holding down a key on my keyboard then pressing any of my extra button (not LMB/RMB) on my mouse the keyboard key is interupted and i need to press and hold down again.

Unreleated. Any modifiers from the keyboard Shift/Ctrl/Alt dont work with my mouse extra button keys.  Shift + Extra Button 1  when typing should give me ! instead i just get 1 as if it doesnt recognized the two imputs at the same time.

For Reference i have a Razer Naga. The Extra Buttons are default mapped to the 1,2,3... above qwerty.

hrvach, 2025-01-28:
I think your mouse extra buttons are acting like a keyboard, so when the message arrives from e.g. board B, it's sent to the OS individually and not combined with the existing keyboard state. Normally the OS would take care of this as it would consider these to be two individual devices, but since these are sent under the same device to the OS, it thinks you released one key and pressed another.

I'll leave this open as a feature request to support 2 keyboards simultaneously and allow state merging.

TheRedfoox, 2025-04-19:
What is currently feasible to overcome this? Typically when on Valorant I hold "W" to move forward and press "8" of my Razer Naga it stops my movement. But on the other hand, using mouse 5/4 (previous/next) does not pose problems, are we talking about the same thing?

ahshift, 2025-04-21:
Yes its treating both inputs as if its comming from one keyboard. For example is you try to hold W and press 8 on the same keyboard it will interupt the held W (Ex. WWWWWWWWWWW8 then nothing until you press and hold W again). If you had two keyboards plugged into your computer and held w on one while pressing 8 on the other it would continue to keep the held W input while injecting the 8 (Ex: WWWWWWWWW8WWWWWWW). No fix for this yet as im way to far from this kind of coding to help out.

ahshift, 2025-04-21:
I think maybe i fully dont understand how all perihpials work but you're correct. on the same keyboard while typing looks interupted a game can handle holding a key and taping another while still not interupting the hold. Im guessing its all related to how the keyboard style inputs are handled from the second source (mouse) and mixing with the keyboard input.

TheRedfoox, 2025-04-21:
My theory (without knowing exactly how your code works perfectly) would be that the HID report emits by the mouse therefore has keyboard keys that overwrite the state that the card sends from the keyboard. The "W" key is not present as if pressed on the information transmitted by the keyboard keys with the mouse.

I'm starting to understand peripherals/HID so be indulgent 😅

ahshift, 2025-08-25:
Just wanted to update this as i think i found a better way to describe and have you be able to recreate the "issue"/freature request.

If you plug in two keyboards instead of mouse and keyboard, they "interupt" each other. 

Normal/No Deskhop, Plug in two keyboards into your computer. You can press shift on one board and any key on the other board and it will apply the shit to that keystroke.  So holding SHIFT on Keyboard one + Pressing s on Keyboard 2 will return a captial S.

With Deskhop. With two keyboards plugged in, Pressing/Holding Shift on one keyboard and any key on the other returns just the unshifted key value from the second keyboard.

What this looks like, the "issue/feature" is when in a game holding down a movement key, pressing a key on the other keyboard (extension of mouse) stops all key commands from the keyboard as well as unable to use a shift+keystroke between the two devices.

ahshift, 2025-08-25:
This was tested with the latest firmware .73 just to be sure it wasnt resovled, dont want to be wasting any time. Please let me know if theres any debugging/more data i can collect to help with this.

closed 2025-10-06 by ahshift