hrvach/deskhop #215: Input delay with logitech wireless receiver

nathank07, 2025-01-26:
I recently bought a Logitech G Pro Superlight 2 to replace my existing G Pro Wireless mouse and unfortunately it seems like there is a significant delay on the superlight receiver (but not a direct cable connection). The input delay occurs regardless on if a keyboard is plugged in or not.

Some potentially useful info:

Sonoma 14.2.1
KDE Plasma 5.24.7 (Pop_OS! 22.04)
Deskhop v0.70

Superlight 2 connected by cable:
```
$ sudo usbhid-dump -a 1:3
001:003:002:DESCRIPTOR         1737865238.535882
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0

001:003:001:DESCRIPTOR         1737865238.540873
 05 01 09 06 A1 01 85 01 95 08 75 01 15 00 25 01
 05 07 19 E0 29 E7 81 02 81 03 95 05 05 08 19 01
 29 05 91 02 95 01 75 03 91 03 95 06 75 08 15 00
 26 FF 00 05 07 19 00 2A FF 00 81 00 C0 05 0C 09
 01 A1 01 85 03 95 02 75 10 15 01 26 FF 02 19 01
 2A FF 02 81 00 C0 05 01 09 80 A1 01 85 04 95 01
 75 02 15 01 25 03 09 82 09 81 09 83 81 00 75 01
 15 00 25 01 09 9B 81 06 75 05 81 03 C0

001:003:000:DESCRIPTOR         1737865238.545876
 05 01 09 02 A1 01 09 01 A1 00 95 10 75 01 15 00
 25 01 05 09 19 01 29 10 81 02 95 02 75 10 16 01
 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01 75 08
 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38 02 81
 06 C0 06 00 FF 09 F1 75 08 95 05 15 00 26 FF 00
 81 00 C0
```

Superlight 2 connected through receiver:
```
$ sudo usbhid-dump -a 1:4
001:004:002:DESCRIPTOR         1737865318.899918
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0

001:004:001:DESCRIPTOR         1737865318.904910
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 95 05 05 08 19 01 29 05
 91 02 95 01 75 03 91 03 95 70 75 01 05 07 19 04
 29 73 81 02 95 05 19 87 29 8B 81 02 95 03 19 90
 29 92 81 02 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 06 81 03 C0

001:004:000:DESCRIPTOR         1737865318.909916
 05 01 09 02 A1 01 09 01 A1 00 95 10 75 01 15 00
 25 01 05 09 19 01 29 10 81 02 95 02 75 10 16 01
 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01 75 08
 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38 02 81
 06 C0 06 00 FF 09 F1 75 08 95 05 15 00 26 FF 00
 81 00 C0
```

hrvach, 2025-01-28:
Hmm, maybe the polling frequencies differ significantly, filling up the buffer. Is the mouse cursor super super slow, lagging up to 2-3 seconds and basically unusable unless you make very small, infrequent movements?

nathank07, 2025-01-29:
No, the mouse functions as expected it just has a small delay (I would say closer to 500ms), I did notice on Deskhop v0.61 that when using the keyboard it would occasionally freeze up and repeat keys but that behavior no longer seems prevalent.

hrvach, 2025-01-29:
That's strange, like the buffer is constantly full. It would be a good test to make a keyboard shortcut that wipes the mouse buffer, I wonder if that would fix the latency.

kkourin, 2025-04-05:
Could confirm this with my own Superlight 2. I got the impression that the delay was in the receiver, not the deskhop. Seemed to me like clicking did not have the same delay. I ended up switching to a different wireless mouse which did not have this issue.