hrvach/deskhop #216: Keyboard keys alphabetically shifted

SyedAmerGilani, 2025-01-27:
When using my Keyboardio Model 100, I encountered an interesting issue. With the initial versions of Deskhop, everything functioned well except for the media keys, which weren't operational. However, after upgrading to the current version, I noticed a peculiar behavior: all keys are now shifted alphabetically by 5 positions. For instance, when I type "abcdef", the output shows as "efghijk". 

To rule out any hardware issues, I've conducted several tests. The keyboard works perfectly fine without Deskhop running, and when I connect a different keyboard while using Deskhop, everything operates normally. This shifting behavior persists across both my connected computers - one running Linux and the other Windows - which suggests the issue is specifically related to the interaction between the current Deskhop version and my Keyboardio Model 100.

Here is the output of usbhid-dump
```
007:009:004:DESCRIPTOR         1737971532.139375
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 08
 15 00 25 01 95 08 75 01 81 02 05 01 09 30 09 31
 16 00 00 26 FF 7F 75 10 95 02 81 02 09 38 15 81
 25 7F 75 08 95 01 81 06 C0 C0

007:009:003:DESCRIPTOR         1737971532.142369
 05 0C 09 01 A1 01 85 04 15 00 26 FF 03 19 00 2A
 FF 03 95 04 75 10 81 00 C0 05 01 09 80 A1 01 85
 05 15 00 26 FF 00 19 00 29 FF 95 01 75 08 81 00
 C0 05 01 09 02 A1 01 09 01 A1 00 85 01 05 09 19
 01 29 08 15 00 25 01 75 01 95 08 81 02 05 01 09
 30 09 31 09 38 15 81 25 7F 75 08 95 03 81 06 05
 0C 0A 38 02 95 01 81 06 C0 C0

007:009:002:DESCRIPTOR         1737971532.145368
 05 01 09 06 A1 01 05 08 19 01 29 08 15 00 25 01
 75 01 95 08 91 02 05 07 19 E0 29 E7 81 02 75 08
 95 07 81 01 75 01 95 04 81 01 19 04 29 DD 95 DA
 81 02 95 02 81 01 C0
```

hrvach, 2025-01-28:
I think I know what's going on here - thanks for the hid dump. 

    Usage Minimum (KB A),           ; Keyboard a and A (04h, selector)
    Usage Maximum (KP Hexadecimal), ; Keypad Hexadecimal (DDh, selector)
    Report Count (218),

Most keyboards seem to do usage minimum 0 and I'm guessing the nkro parsing routine handles this incorrectly.

If I try providing a fix, would you be able to test?

SyedAmerGilani, 2025-01-29:
That makes sense. Yes I could test the fix.

hrvach, 2025-01-29:
When you find a moment, could you also do usbhid-dump -e stream (while having keyboard plugged directly to the pc) and type "abc" then share?

SyedAmerGilani, 2025-01-30:
Sure here it is:
```
007:006:002:STREAM             1738218293.550271
 00 00 04 00 00 00 00 00 10 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

007:006:002:STREAM             1738218293.663044
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

007:006:002:STREAM             1738218294.664354
 00 00 05 00 00 00 00 00 20 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

007:006:002:STREAM             1738218294.780312
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

007:006:002:STREAM             1738218295.613097
 00 00 06 00 00 00 00 00 40 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

007:006:002:STREAM             1738218295.720381
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00
```

hrvach, 2025-02-01:
Does this perhaps work better? Thanks for testing.

[deskhop-test-nkrofix.zip](https://github.com/user-attachments/files/18630602/deskhop-test-nkrofix.zip)

SyedAmerGilani, 2025-02-03:
This solves the Bug. Nothing is shifted anymore. Thank you!

closed 2025-02-03 by SyedAmerGilani