hrvach/deskhop #218: Crazy mouse movement with Kensington Expert Mouse

rygwdn, 2025-01-29:
I'm getting erratic movement & clicking when I try to use my Kensington Expert Mouse (a trackball). It looks like it shows up as two different devices based on the usbhid-dump output. I wonder if that could be part of the issue?

Here are the descriptors:

<details><summary>Mouse</summary>
<p>

```cpp
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x95, 0x05,        //     Report Count (5)
0x75, 0x01,        //     Report Size (1)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x05,        //     Usage Maximum (0x05)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x75, 0x03,        //     Report Size (3)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0C,        //     Usage Page (Consumer)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x95, 0x01,        //     Report Count (1)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Consumer Control)
0xA1, 0x00,        //   Collection (Physical)
0x75, 0x0C,        //     Report Size (12)
0x95, 0x02,        //     Report Count (2)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x16, 0x01, 0xF8,  //     Logical Minimum (-2047)
0x26, 0xFF, 0x07,  //     Logical Maximum (2047)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x01,        //   Report Count (1)
0x09, 0xCD,        //   Usage (Play/Pause)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x0A, 0x83, 0x01,  //   Usage (AL Consumer Control Configuration)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0xB5,        //   Usage (Scan Next Track)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0xB6,        //   Usage (Scan Previous Track)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0xEA,        //   Usage (Volume Decrement)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0xE9,        //   Usage (Volume Increment)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
[... 928 lines cut ...]
001:009:000:STREAM             1738155795.459137
 20 64 8B 00

001:009:000:STREAM             1738155795.531106
 21 9F 08

001:009:000:STREAM             1738155795.603140
 22 09

001:009:000:STREAM             1738155795.683141
 23 79 00 79 00

001:009:000:STREAM             1738155795.755135
 25 13

001:009:000:STREAM             1738155795.827098
 28 00

001:009:000:STREAM             1738155795.891123
 29 00

```

</p>
</details>

rygwdn, 2025-01-29:
for sure, I'm happy to test!

chri2, 2025-01-30:
Could this be related to https://github.com/hrvach/deskhop/issues/133 ?

If so: I'm replacing (just ordered) the not working mouse from https://github.com/hrvach/deskhop/issues/133 by the older, working version. If it helps I could contribute the not-working version for testing if interest be.

hrvach, 2025-02-02:
> for sure, I'm happy to test!

Any changes with this perhaps?

[deskhop-test-mouseids.zip](https://github.com/user-attachments/files/18630685/deskhop-test-mouseids.zip)

hrvach, 2025-02-02:
> Could this be related to [#133](https://github.com/hrvach/deskhop/issues/133) ?
>
> If so: I'm replacing (just ordered) the not working mouse from [#133](https://github.com/hrvach/deskhop/issues/133) [...]

Yes, this is the same issue. Please try the zip as well, let me know if it behaves better?

rygwdn, 2025-02-02:
That didn't work unfortunately, but I took a stab at a fix myself in #219 which does seem to be working for me!

chri2, 2025-02-02:
> [deskhop-test-mouseids.zip](https://github.com/user-attachments/files/18630685/deskhop-test-mouseids.zip)
> 
> Probably doesn't work, but who knows :)

The behavior for Cherry MW 8C Advanced USB-C is the same: Doesn't work.

> That didn't work unfortunately, but I took a stab at a fix myself in [#219](https://github.com/hrvach/deskhop/pull/219) which does seem to be working for me!

I also compiled the version by @rygwdn (I hope I did everything correctly, I downloaded the code from [here](https://github.com/hrvach/deskhop/tree/32305f1a6a7ea40623ad1cc31cf84be972a52cea) as this [zip](https://github.com/hrvach/deskhop/archive/32305f1a6a7ea40623ad1cc31cf84be972a52cea.zip) and it seemed to contain the patches. Compiled that.)

It doesn't work for Cherry MW 8C Advanced USB-C.

chri2, 2025-02-02:
> If it helps I could contribute the not-working version for testing if interest be.

The carrier announced that my replacement Cherry MW 8C Advanced **USB-micro** will arrive on monday. If everything is fine I could send of one of my replaced **usb-c** versions for testing during the week.

rygwdn, 2025-02-02:
@chri2 not sure how you're flashing yours, but I couldn't use the method using the keyboard and instead had to directly flash the two halves separately because the version number isn't bumped

chri2, 2025-02-02:
> [@chri2](https://github.com/chri2) not sure how you're flashing yours, but I couldn't use the method using the keyboard and instead had to directly flash the two halves separately because the version number isn't bumped

Uh, I used the keyboard method. I'll try tomorrow with a direct flash and let you know about the result - thanks for the information!

hrvach, 2025-02-07:
> That didn't work unfortunately, but I took a stab at a fix myself in [#219](https://github.com/hrvach/deskhop/pull/219) which does seem to be working for me!

@rygwdn could you please be so kind to try if

[deskhop-test-mouseids2.zip](https://github.com/user-attachments/files/18699217/deskhop-test-mouseids2.zip)

still works to solve your issue? I'm trying to do some refactoring but have nothing to test the issue for possible regression. Thanks!

hrvach, 2025-02-09:
Please try v0.71 pre-release, does the issue still persist?

rygwdn, 2025-02-13:
I can confirm that 0.71 works!

rygwdn, 2025-02-13:
actually, I spoke too soon. It looks like it breaks the ability to hold the mouse buttons down! this fixes it though:
```diff
diff --git i/src/mouse.c w/src/mouse.c
index a46962a..641c9a2 100644
--- i/src/mouse.c
+++ w/src/mouse.c
@@ -219,12 +219,13 @@ void do_screen_switch(device_t *state, int direction) {
         switch_virtual_desktop(state, output, output->screen_index + 1, direction);
 }
 
-inline void extract_value(bool uses_id, int32_t *dst, report_val_t *src, uint8_t *raw_report, int len) {
+static inline bool extract_value(bool uses_id, int32_t *dst, report_val_t *src, uint8_t *raw_report, int len) {
     /* If HID Report ID is used, the report is prefixed by the report ID so we have to move by 1 byte */
     if (uses_id && (*raw_report++ != src->report_id))
-        return;
+        return false;
 
     *dst = get_report_value(raw_report, len, src);
+    return true;
 }
 
 void extract_report_values(uint8_t *raw_report, int len, device_t *state, mouse_values_t *values, hid_interface_t *iface) {
@@ -246,7 +247,10 @@ void extract_report_values(uint8_t *raw_report, int len, device_t *state, mouse_
     extract_value(uses_id, &values->move_y, &mouse->move_y, raw_report, len);
     extract_value(uses_id, &values->wheel, &mouse->wheel, raw_report, len);
     extract_value(uses_id, &values->pan, &mouse->pan, raw_report, len);
-    extract_value(uses_id, &values->buttons, &mouse->buttons, raw_report, len);
+
+    if (!extract_value(uses_id, &values->buttons, &mouse->buttons, raw_report, len)) {
+        values->buttons = state->mouse_buttons;
+    }
 }
 
 mouse_report_t create_mouse_report(device_t *state, mouse_values_t *values) {
```

rygwdn, 2025-02-14:
I opened a PR with the changes too https://github.com/hrvach/deskhop/pull/225

hrvach, 2025-02-25:
> I opened a PR with the changes too [#225](https://github.com/hrvach/deskhop/pull/225)

Thank you, will merge after testing - a bit slow due to work travel, apologies for not getting to this earlier.

rygwdn, 2025-03-13:
with the latest builds this is working perfectly. Thank you!

closed 2025-03-13 by rygwdn