hrvach/deskhop #220: Scrolling using trackball causes cursor to jump and scrolling to stop

Joonetic, 2025-02-06:
Hi there,

When using the trackball for scrolling (set to hold a button then scroll) the cursor seems to jump and scrolling stops after a few scrolling steps. I then have to let go of the button modifier and start scrolling again to move up and down the page. Hope that makes sense. Any settings in the config I can play with to fix this? Running .70

hrvach, 2025-02-07:
You should be able to fast-switch using the switching hotkey still, but I'm not sure I understand the issue (it's been a while since I've seen a trackball). There have been some recent patches and fixes that might be relevant to this behavior. I'll be releasing a new version shortly so could you please try that one first once it's available?

Joonetic, 2025-02-07:
> You should be able to fast-switch using the switching hotkey still, but I'm not sure I understand the issue (it's been a while since I've seen a trackball). There have been some recent patches and [...]

Hi sorry I'll try to explain it better. I use mac mouse fix to set up the trackball to scroll if I hold down the back button on my mouse. If I do that the cursor freezes and let's me scroll but it also creates this second cursor that starts to move up and down depending on which way I'm scrolling. Once that "2nd" cursor reaches the top of my monitor I am unable to scroll further until I release the buttons and try to scroll again. Hopefully this helps? If not i can try to capture a video of it.

hrvach, 2025-02-09:
Please try v0.71 pre-release, does the issue still persist?