hrvach/deskhop #223: use two USB devices， How can this be achieved in the code?

zoushiyin, 2025-02-13:
Hello,  

I would like to use two USB devices (a mouse and a keyboard) simultaneously on a single RP2040, using GPIO pins 2, 3 for one device and GPIO pins 4, 5 for the other. How can this be achieved in the code?

hrvach, 2025-02-13:
I don't think it can, since you don't have enough PIO state machines. Maybe a USB hub IC would work, but hub support is not spectacular so that might or might not work OK.

zoushiyin, 2025-02-13:
Based on the information provided in this GitHub issue #65, the setup should work. However, I have not been able to achieve successful testing despite my efforts.

zoushiyin, 2025-02-15:
> I don't think it can, since you don't have enough PIO state machines. Maybe a USB hub IC would work, but hub support is not spectacular so that might or might not work OK.

Hi @hrvach ,

Thank you for your previous guidance.

Given the scenario where using two sets of GPIO simultaneously isn't feasible, I'm considering implementing a USB hub solution to expand the external connections using a USB hub IC. However, I'm uncertain about the necessary modifications in tusb_config.h or setup.c to accommodate this setup.

Could you please advise on what specific changes need to be made within these files to properly support an external USB hub? Specifically, are there configurations related to USB port management or device enumeration that should be adjusted?

Additionally, if there are any recommended USB hub ICs known to work well with this project or any existing examples of similar setups, it would be greatly appreciated if you could share them.

Thank you very much for your help and support.

mike-o-tronic, 2025-02-16:
Hi @zoushiyin ,
Hi @hrvach ,

I've been using this awesome project with a an Anker portable USB hub with standard USB mouse and keyboard as well as with Logitech Unifying receiver with wireless mouse and keyboard on a single board without any issues. My boards are running with updated pico-sdk and Pico-PIO-USB library, but I'm not sure if that brought any improvements with hub combability. The update was done before I started using the hub.

Also, this shouldn't have any affect on the combability, but I'm using a slightly different hardware setup. I got the Feather RP2040 USB Host board, as it was more convenient with USB-C and the USB host onboard. A pull with request with the library update, support for the Feather board and an additional feature was just submitted. And I'm sorry for it being such an extensive request. I hope I did at least the commits well enough that you can cherry pick if you want to take over some changes, @hrvach .

Cheers
Mike

zoushiyin, 2025-02-17:
Hi @mike-o-tronic,

Thank you for sharing your experience and insights.

I was wondering if your version of DeskHop https://github.com/mike-o-tronic/deskhop has been updated to the latest pico-sdk and Pico-PIO-USB libraries. If so, would it be possible for me to download and test your version to see if it resolves the issues I'm encountering?

Any guidance or pointers on how to proceed would be greatly appreciated.

Thank you again for your help and support.

zoushiyin, 2025-02-18:
Hi @mike-o-tronic ,

Hello! I noticed that you mentioned updating to the latest pico-sdk and Pico-PIO-USB libraries. Could you please share the specific modifications you made before compiling and flashing the firmware? I am also interested in upgrading to use multiple USB devices simultaneously.

Specifically:

What changes did you make to the configuration files (e.g., tusb_config.h, setup.c)?
Were there any additional steps or configurations required to ensure compatibility with multiple USB devices?
Are there any particular versions of pico-sdk and Pico-PIO-USB that you recommend?
Any guidance or detailed steps would be greatly appreciated, as I would like to achieve similar functionality with my setup.

Thank you very much for your help and support.

mike-o-tronic, 2025-02-23:
@zoushiyin 

The libntimeout branch includes the updates. But nothing was changed in the USB configuration. The only changes are the library updates, a timeout function, and added compatibility to the board I'm using. You can review the [commits](https://github.com/mike-o-tronic/deskhop/commits/libntimeout) if you like.

The versions I'm using are:
pico-sdk 2.1.0
Pico-PIO-USB 0.6.1

hrvach, 2025-02-25:
Thanks for the contribution @mike-o-tronic, apologies for the delay - I'm traveling for work but will take a look as soon as I'm back.