hrvach/deskhop #224: Keyboard and Mouse Not Switching Seamlessly

apsessoms, 2025-02-13:
Hello and thank you for providing the community with such an amazing product. My co-worker told me about this and I ordered it immediately. I just need some finishing touches to make it work! :)

I recently purchased the DeskHop-Fast Desktop Switching-Single PCB V1.0 seamlessly switch between my desktop and laptop using a single keyboard and mouse. However, I’m running into an issue where the switching behavior isn't working as expected.

Here’s what I’m experiencing:

- My keyboard works on both my desktop and laptop, but I have to manually press the Computer 1/Computer 2 keys on my Logitech keyboard to switch between them.
- My mouse does not work on my desktop, but the keyboard does.
- My mouse works on my laptop, but the keyboard does not unless I manually toggle between systems using the Logitech keyboard keys.

It seems like DeskHop isn’t recognizing the automatic transition between systems as intended, possibly due to how the Logitech keyboard handles multi-device switching. 

Do you have any suggestions on how to troubleshoot or configure DeskHop to resolve this issue? Let me know if you need any additional details!

I have tried to:

- deploy the initial firmware that was uploaded to this repo 4 days to both pico boards
- update the raspbery pico firmware
- read through alll the issues on this github

Thanks for any help!

hrvach, 2025-02-25:
Apologies for the delayed response, work is a bit overwhelming lately. Can you try another "plain" wired keyboard and mouse for starters, to rule out hw is faulty? I'm not sure the logitech multi-device concept is compatible with this, you shouldn't be required to switch anything on the logitech but simply plug both computers as the output and the keyboard as the input.

apsessoms, 2025-03-11:
it was my cable. one was data. one was not. 🤣

hrvach, 2025-03-18:
Thanks for providing this data point and behavior description, I never thought to blame the cable. Might put this in the wiki as symptoms + possible reason. Glad to hear it's working for you!

closed 2025-03-18 by hrvach